opentx/opentx #7582: Taranis D9X Plus SE

jetpr101, 2020-05-04:
Hi 
Can the scroll button be less sensitibe this is hard to works with it.

thanks

kilrah, 2020-05-04:
One click is one click, it's how the hardware is...

closed 2020-05-04 by kilrah

mpjf01, 2020-05-05:
When using the roller it can be difficult to get the selection to settle on whatever you are trying to choose and therefore it's way to easy to inadvertently select the wrong item. The selection seems to move either up or down easily, particularly at the point when you click the roller to choose it. 

Would it be possible to program a tiny delay so that when you click the roller the movement part doesn't work for that time, or would that be too annoying for other aspects of the roller use?