opentx/opentx #7586: Jumper T16 Pro V2 Internal Radio off

Jannovik, 2020-05-06:
Hello,
I updated the Jumper T16 Pro V2 (2020 Model) to firmware 2.3.7 and I can't turn on the internal radio. The option is not available in the radio. In the OpenTX Companion I see the internal radio options, but I click to turn it on and only off shows up in the drop down menu. I flashed to firmware 2.3.5 and it works again.
**General information**
 - OpenTX version 2.3.7
 - Hardware T16 Pro V2
 - OpenTX Companion 2.3.7 macOS 10.15.4

3djc, 2020-05-06:
You forgot to enable internalmulti before downloading the firmware

closed 2020-05-06 by 3djc

Jannovik, 2020-05-06:
How do I do that?

YaBoiWhiteRice, 2020-05-10:
Hello @Jannovik, How did you get back to 2.3.5 after you upgraded to 2.3.7? I have upgraded and I have had some problems and would like to go back, but cannot get a 2.3.5 firmware file.

Jannovik, 2020-05-10:
What problems are you having? I made a backup of the firmware before upgrading. I had many problems, but they ended up being hardware related. I ended up emailing Jumper RC and they helped me fix it.

YaBoiWhiteRice, 2020-05-12:
I was having “key stuck” and it wasn’t taking any input. I worked with people in the Discord server. I ended up swapping the ribbons for the scroll wheel and right button set. That fixed it up nicely. Thank you though. I should really start making backups of my models and versions.

Jannovik, 2020-05-12:
I had a similar problem. The SH switch would trigger 2 other switches. I also swapped the ribbons and that fixed it. Good luck and happy flying.

YaBoiWhiteRice, 2020-05-12:
Thanks, you too!