opentx/opentx #7591: QX7 dies after upgrade to 2.3.7

nmw01223, 2020-05-08:
QX7 was new in Oct '17, came with 2.2.0, and has never been updated. Never had any problems. Used as RF transmitter and also simulator controller (PC via USB). SD card fitted. Runs on 6x AA NiMH. Decided to update code 5 days ago to get multimodule support (not in that original 2.2.0 build).

Installed companion 2.3.7, downloaded and installed 2.3.7 code (lua, noheli), booted up, then from the code updated the boot code. Models were auto-updated, all fine.

Then, this morning, switch on, and just got a click. Initially assumed left on and flat batteries, but not so - nearly full charge. Recharged anyway, and after hunting around for help reprogrammed the 2.3.7 code in boot mode from companion via USB (no batteries). All OK again.

Using it this morning and suddenly it came up with low (Tx) battery warning (5.4V). Powered it off, and again, would not boot. Batteries had been fully charged yesterday, and all were all still OK. Repeated reprogramming as above - all OK again. Batteries read 8.1V.

Significant points may be:

1. Flew OK outside in the last few days, but battery was never that high (started on Monday around 7.7V and went down to ~7.1V, hence recharge yesterday).
2. Yesterday, after recharge, was using it with a simulator via USB. It was unplugged in the usual way at the end.
3. To-day, also being used as simulator controller via USB, and the low warning was in the middle of use.
4. In neither case could get into boot mode, either.
5. Reprogramming main code (not boot code) seems to sort it.

Wondering if (a) some code  or battery multiplier somehow got corrupted when plugged in via USB, or (b) although it seems bizarre, something goes wrong when battery voltage drops past around 8.2V.

Currently have it on, and running with a simulator, to try to repeat it with simulator. Can find no similar issue in open/closed bug lists.

bigtwin, 2020-05-25:
Just to add this to the topic - I have updated my QX7 from 2.2.4 to 2.3.7 just yesterday. No issues like that. 
You havent mentioned that you updated the SD card content ...(?)

nmw01223, 2020-05-27:
Since my original report, have to admit it hasn't gone wrong .... .

Wondering if it was much simpler - poor battery connection, and voltage really did drop. Reprogramming may have solved it simply because the battery got disconnected / reconnected in the process.

If you use 6x AA batteries (as I do), they are gripped very tightly by the holder, and just maybe therefore, the spring at one end of each cell was unable to push the cell hard enough against  the other end to make reliable contact.

On that basis therefore, aiming to close this, and will open something else if it later proves to be s/w related.

closed 2020-05-27 by nmw01223