opentx/opentx #7593: Taranis X10S radio in OpenTX issues

04fxdwgi, 2020-05-09:
Radio: Taranis X10S
OpenTX rev: opentx-x10-lua-ppmus-en-2.3.7
OpenTX Companion rev: 2.3.7
Issues:
1. Edit Radio Settings / hardware tab: Shows two components EX1 and EX2. Unknown what these 2 are
2. Edit Radio Settings / Calibrations tab: Shows calibration numbers for EX1 and EX2
3. Edit Radio Settings / Calibrations tab: Shows calibration numbers for components ??? and ??? on last two lines. Unknown what ??? and ??? are.

[image]
[image]

kilrah, 2020-05-09:
EX1/EX2 are additional pots that can be connected to free connectors on the radio board.
Please note that this is not a place for user questions.

04fxdwgi, 2020-05-09:
Andrea, 

 

Thank you for the rapid response.

 

Please excuse my ignorance, as this is first time on GitHub. My train of thought was that it went hand in hand with the Hardware Calibration for the two slots labeled ???

 

Are the two calibrations labeled ??? a mislabel? I was unaware X10S had available inputs (EX1 and EX2) for unused POTs.

 

Thank you again

 

Regards

Bill Curtis

kilrah, 2020-05-09:
The ??? are nothing, likely we just display too many lines.

04fxdwgi, 2020-05-09:
Thanx again

closed 2020-09-04 by elecpower