opentx/opentx #7594: Critical watchdog reset on 9xrPRO and 2.3.7 version

blairfancy, 2020-05-09:
**Describe the bug**
Intermittent watchdog reset when navigation on model menu. Very critical since it can occurred when on flight.

**To Reproduce**
Steps to reproduce the behavior:
1. Go model setup
2. Play with option ( especilay extern mode PPM,DSM...) 
3. radio reset and "!" appear on top right of radio

**Expected behavior**
Expect relaible radio and no Watchdog at all

**Screenshots**
NA

**General information**
 - OpenTX version latest stable 2.3.7 for 9xrPRO
 - Hardware 9xrPRO without module

pfeerick, 2020-05-10:
I had a few pokes and prods around the `External RF` modes, changing PPM and DSM2 stuff to all sorts of wacky values, but haven't had this yet. I do have a FrSky XJT module in the Tx though, if that somehow changes things.

blairfancy, 2020-05-10:
How can i help on that? Any test to be done or new beta version to try?

dgatf, 2020-05-29:
Why would you change the module settings during flight?

blairfancy, 2020-05-29:
@dgatf issue does not occur only when you change module setting the same can occure when you navigate through other setup like curve or mixer.

pfeerick, 2020-05-30:
Because you might be fine-tuning settings. 🤦

closed 2021-03-20 by stale[bot]

BrettRD, 2021-05-07:
This is not fixed at 2.3.11 on 9xr-pro.

To reproduce, select a model configured with ppm, select ppm and wait five seconds with flashing cursor.
Found while reproducing https://github.com/opentx/opentx/issues/7772