opentx/opentx #7595: Add "power smart port" option for powering receivers outside the quad (for binding)

jag-88, 2020-05-09:
It's sometimes frustrating binding a receiver after installing it in the quad.

I would like the ability to power a receiver off the smart port pins outside of the flashing process. This could be very useful for binding/registering receivers outside of the quad without needing to purchase a laboratory power supply.

The smart port already powers the receiver during the flashing process so the ability to get power from the smart port pins is already there, I propose adding a selection under radio setup to turn on the power to the smart port pins so that you can bind the receiver using that power source.

3djc, 2020-05-09:
Already existing

closed 2020-05-09 by 3djc

jag-88, 2020-05-09:
@3djc The feature exists or the feature request already exists? If the feature exists could you tell me where it is in the menu?

3djc, 2020-05-09:
Exist since March 10, in Radio/hardware menu, almost at the end

jag-88, 2020-05-09:
Found it! Thank you so much! :)