opentx/opentx #7600: Selection of any internal module mode on X9E transmitter crashes the screen

FS4000, 2020-05-12:
Describe the bug
Selection of any internal module mode on transmitter crashes the screen

To Reproduce
Steps to reproduce the behavior:

1. Make a new model (MODELL44), with or without wizard on X9E
[image]
2. Click on Menu and select on the second screen INTERNAL HF- MODULE > MODE
[image]
3. Selection of any internal module mode crashes this screen and brings
    it back to the first screen. Its not possible to choose the internal module.
    On the first screen an exclamation mark appears on the upper right corner.
[image]
4. Selection of another model (MODELL43) can not bring the transmitter back to normal function.
    Only a restart is the solution.
[image]
5. On Companion, i can choose the internal module and it is active on the transmitter.
    I found two Issues with similarly problems: #7126 and #7594
    The same configuration on my Taranis QX7 works as it should, no problems!
 
Expected behavior
This reaction maybe dangerous and should not happen when running the system on the field.

General information
 - OpenTX version (2.3.7)
 - Hardware (Frsky X9E XJT ACCST with IRangeX Multiprotocol Module, Firmware v1.3.0.95) 
 - OpenTX Companion (2.3 on Windows 10)
 
Thank you for your work, i like OpenTX!

Kind regards, Uwe

3djc, 2020-05-12:
Could you include you OTX file ? It is clearly not happening here, most likely, you have set using companion a forbiden configuration.

FS4000, 2020-05-12:
Here my otx- testfile (scroll down to MODELL43). I made the models not in companion, they were created on the transmitter first! On my older configurations it also works, but i can't tell when the error occurred for the first time.
[X9E_Test.otx.txt](https://github.com/opentx/opentx/files/4617701/X9E_Test.otx.txt)

Rivelino911, 2020-05-12:
Same for me on Taranis X9D+ (non 2019) when i select internal module. For external no problem.
After i write 2.3.7 also removed SD and made a factory defaults. No go.

3djc, 2020-05-14:
> Here my otx- testfile (scroll down to MODELL43). I made the models not in companion, they were created on the transmitter first! On my older configurations it also works, but i can't tell when the [...]

This file is empty

FS4000, 2020-05-14:
Hello 3djc,
please rename the file (delete .txt) and then in Companion scroll down to MODELL43.

3djc, 2020-05-14:
Can duplicate the issue, please give us an eeprom backup instead

FS4000, 2020-05-14:
Here the Eeprom file:
[eeprom-2020-05-14-175520.bin.txt](https://github.com/opentx/opentx/files/4629532/eeprom-2020-05-14-175520.bin.txt)

FS4000, 2020-05-14:
If the screen crashes, as a result of the module selection,
the voltage display shows zero volts for a short moment.
Maybe this will help.
Regards, Uwe

bsongis, 2020-06-04:
Did you all select the "internalppm" option in Companion? If yes the bug is fixed (and the option removed as it shouldn't be used here!)

closed 2020-06-04 by bsongis