opentx/opentx #7603: 2.3.8 - Para Bluetooth on X12S does not come out in telemetry mode with the assigned name

robustini, 2020-05-14:
I installed the Para Bluetooth in my X12S, I had previously also installed the ISRM.
This BT module with the latest code does not come out in telemetry mode with the assigned name, I find it as "Hello" on my smartphone, when instead it should find it as "x12smarco", the name I have assigned.
The "trainer" function works without problems, tested with Jumper T16, on which I had installed the original BT module of the X12S (not PARA).

**General information**
 - OpenTX  V2.3.8 N221 (e.g. 2.1.23 or N123)
 - Hardware: Horus X12S with ISRM and Para Bluetooth

robustini, 2020-05-16:
I've found a similar issue and I checked, it does the same thing on my X10S Express too.

https://github.com/opentx/opentx/issues/7381

kilrah, 2020-05-16:
Believe that one's been in the hands of FrSky for fixes to the module's firmware for months/years...

robustini, 2020-06-03:
@bsongis please could you inquire from FrSky if something can be done?

robustini, 2020-06-13:
However I tried the Trainer mode with between five tx, two X10S Express, one X12S with PARA Bluetooth and two T16, the link is painful and unusable, after less than a meter is lost.
I don't know if it is a problem of internal parameterization of the firmware of the Bluetooth module but it is unusable.

kilrah, 2020-06-13:
Did you buy a FrSky PARA module to put in your T16s? They have custom firmware so of course it wouldn't work if you use generic modules.

robustini, 2020-06-13:
> 
> 
> Did you buy a FrSky PARA module to put in your T16s? They have custom firmware so of course it wouldn't work properly if you use generic modules.

No Andre of course, on the T16 I used the BT V1 what was originally on my X12S (upgraded with the new PARA BT).
PARA and not PARA they link without problems, but with a ridiculous range, it is unusable.
I don't know if it's a question of BT firmware or of too low power, but so it doesn't work as it should.
Having 5 tx I tried different combinations, even PARA<->PARA is unusable.

kilrah, 2020-06-13:
Then of course it won't work, the firmware/protocol has changed, that was the whole point of PARA since the previous modules had terrible refresh rate.

PARA works fine here between X9D+2019 and Xlite Pro, at least to about 10m that I can do in my flat....

closed 2021-03-20 by stale[bot]