opentx/opentx #7604: Massive delay on support for Nirvana

evildronepilot, 2020-05-14:
I do appreciate all the hard work has been done by the community but...
Please help me understand something. Why the brand new models of the radio jumper T16 t16s  and t18 and even ratio master ARE supported by open tx almost instantaneously when the Nirvana has been hanging there for almost a year?

3djc, 2020-05-14:
2.4

closed 2020-05-14 by 3djc

evildronepilot, 2020-05-14:
Initially it was 2.3.4 or something like that then moved to next versions and then eventually 2.4 with is it due to released until ,like next year.... For contrast jumper t18 have not even been released yet....

evildronepilot, 2020-05-14:
Seriously?! are you closing it?

3djc, 2020-05-14:
Whats the point of leaving it open ? It is not a bug, work is schedule as per https://www.open-tx.org/radios. Touch won't be supported before 2.4, NV14 requires touch, end of story

evildronepilot, 2020-05-14:
Well, maybe because question remains unanswered? And I'm starting to feel that there is a lot more to the story than that.

calif94577, 2020-05-14:
So as much as I want there to be support for the NV14 and also think there’s less priority for it than others there’s a reason for T16 having almost instant support which is the same reason Frsky lost their crap when it came out. Because it uses the same processor pin outs as one of the other Frsky transmitters that already have support meaning the code did not have to get rewritten. I’m not sure if the T18 is the same way so I can’t speak to that.

Also they are correct that this isn’t a bug or issue so not the right place to post this. But I would love to see the NV14 officially supported. I’m tired of using the UFPV firmware that’s half baked...

evildronepilot, 2020-05-14:
Thank you for actually talking to me and for the advice. 

 I'm not sure if UG firmware is best. At the end of the day it seems to be losing to the flysky branch available here. Also FlySky does not charged extra for the support. 

Ah Politics;) 

Thank you and all thumbs up.

https://github.com/open-flysky

kilrah, 2020-05-14:
The NV14 is a massive pain with its display orientation and touch only controls, the others can be added easily even with pin changes and such but not the NV14.

Thoemse, 2020-05-15:
Guys calm down. OpenTx is free. These fine people do hard work and charge nothing for it. I own  a NV14 too but I don't see myself in a place where I can demand anything. Lets hope for 2.4 and give the devs a break.

ray-nymous, 2023-01-10:
it's 2023 now. and nothing sitll nothing

3djc, 2023-01-10:
Nv14 is supported by EdgeTX