opentx/opentx #7605: [Horus] Pwr On delay missing

F-Po, 2020-05-15:
On OpenTX 2.3.7 the option "Pwr On delay" is missing. The delay seems to be hardcoded to 2s. I don't like to push the button over a long time to power it on. Please add this option or restore the old behavior.

[image]

3djc, 2020-05-15:
What radio/version are you comaring with ?

F-Po, 2020-05-15:
Sorry, I forgot: Horus X12S

3djc, 2020-05-15:
Ok, I understand then. There is not delay added on horus, thats just how the hardware is built, and there is nothing that can be done about that I'm affraid

F-Po, 2020-05-15:
That could be not correct. The behavior was seen by me first in 2.3.5. Until 2.3.3 there was no power on delay.

3djc, 2020-05-15:
Well, I'm running 2.2.4 and I'm need to push 2secs for the radio to turn on, so I must not be looking at what you mean. Could you do a video with 2.3.3 and 2.3.7 showing the difference ?

F-Po, 2020-05-15:
Yes, but it takes a while.

F-Po, 2020-05-15:
Correction: My version before was the 2.3.2.

Here the videos:

OTX 2.3.2: https://youtu.be/X4oUTuV9OvU
OTX 2.3.7: https://youtu.be/6YpfdBtywCw

maru-sama, 2020-05-15:
I just had a quick look at the code, and apparently both Power On Delay and Power Off Delay are defined. But Power ON is neither showing up in companion nor the simulator or the device itself.

3djc, 2020-05-15:
because it is not used by Horus ...

3djc, 2020-05-15:
If another dev wants to have a look, my x10 in 2.3.2 does not behave at all like in the video, so not sure if it is x12 specific, or if my proto has an issue

F-Po, 2020-05-20:
Could someone please check if one of these commits should be the reason for this issue: https://github.com/opentx/opentx/pull/7212
     - The taranis has the power on delay option
     - If currently the same code is used could it be possible that the default/initialisation value of 2s is used also on the Horus?

https://github.com/opentx/opentx/pull/7096
    - Is it possible that currently the Horus always ends in unexpected shutdown and the restore at the next boot will take some time?

3djc, 2020-05-20:
no and no

kilrah, 2020-05-20:
- Horus can't easily have configurable power on delay becasue the data is stored on SD card, and the settings aren't read until a couple of seconds into power up

- X12S was the first of the radios using SD card. No delay was put initially as it had a switch cover.

- Now there are many radios using SD card and no other has a switch cover, so a delay was introduced, and it's there for everything, no interest in maintaining a special case just for X12S

F-Po, 2020-05-20:
Just for clarification: The configurable value is secondary for me. But the issue that I must longpress the power button is primary for me.

F-Po, 2020-11-16:
pls fix!

closed 2020-11-17 by kilrah

F-Po, 2020-11-17:
Is this the way OpenTX-gods work? Kill basic functionalities / create bugs with ass in front and ignore the issues to close! Such a case of arrogance!

F-Po, 2020-11-17:
Oh, inconveniant comments are deleted. I have another one: If you go this way straight, your quality / stability will decrease rapidly in the near future. Today there are simple handling issues like a button that have another behavior, not loaded backgound pics, ... But when this ignorance will reach critical functionality, OpenTX will die rapidly!

Developing new features is always more fun! But fixing issues is essentially **the** main thing in sw development. So fix you f***ing bugs!

kilrah, 2020-11-17:
It's not a bug, it's an intentional change as explained before.

lshems, 2020-11-18:
I do understand his grievance. It's more about how than what. 

But you know me. So ignore.