opentx/opentx #7608: Companion sensor units incorrect in some instances

elecpower, 2020-05-15:
eg Cels unit should be V

[image]

Suspect PR #7529 is the culprit

**General information**
 - OpenTX Companion 2.3.8 Ns on Ubuntu 16.04

kilrah, 2020-05-15:
How did Cels get into the list?
Only discovering it on a real radio will set the correct unit. 
It's known that sometimes a major version upgrade hasn't kept the units right though.

elecpower, 2020-05-16:
Cels was discovered pre 2.3.
The PR was only merged in the last month and added a new unit before volts in Companion. The result a description shuffle of those units below the new unit.

3djc, 2020-05-16:
m/l min is not before volt as far as I can remember, it was last, before hours, but after all "units"

kilrah, 2020-05-16:
It is after volt but before the virtual units, which Cels is one of.

But the linked PR is correct, it's before that that it was wrong and would be unrelated, issue happens while upgrading in eeprom structure handling. 
Did you do the 2.2 to 2.3 upgrade on the radio or on companion?

elecpower, 2020-05-16:
I cannot be certain but more than likely used Companion. Probably would have read radio settings into Companion.
I'll see if I have a backup 2.2 radio eeprom and/or Companion files and try some conversions to see if I can replicate.

elecpower, 2020-05-22:
Investigation so far. Looks like the otx I was using was corrupted by too many tests however reading models and settings from a X9D+ 2.3.7 the unit is blank in Companion logical switches but correctly V on the radio and Companion simulator.
Still working thru the source code but it appears Companion was not updated to reflect the EEPROM version 219 change for 2.3 that added spare sensor units (commit # dcc803c5). PR 7529 added the new unit only.

PR #7653 (2020-05-31): Align companion telemetry sensor units to radio
Fixes #7608 
Mirror sensor units and add offset conversion for eeproms.
Tested on X9D+ 2.1 otx and eepee, and 2.3 otx
closed 2020-05-31 by bsongis