opentx/opentx #7610: Option for Horus to use a Free UART for external Access

brainbubblersbest, 2020-05-17:
I request a Option in companion like internal access. 
In this case for external access. 

Have figure out a simple solution to forward a UART from Stm32 to external Module bay. 
USART TX for pin 1
USART RX for Pin 2

For This it can be used aux_serial (1)
Aux2_serial is still used for Bluetooth serial(USART6) 

The x10 express Pins PB10 and PB11 from Stm32 are used for external access (USART3) . In x12s AND x10s cases this Pins are used for the p13 connector aux serial. From here a trace to jrbay is easy to realise.
This connection is disabled outside a debug OTX Build anyway. 

One cable goes to Jr side r1 solderpad from not used heartbeat pin, from p13 RX
Other goes to pin 1jrbay from p13TX, after cut the trace to pa10 (x10s), or pa15(x12s) 
(rfmboard->mboard connection cable) 

This makes a full conversion (in function) for old non express Horus Devices possible. 
So r9m2019 module with future telemetry improvements are able to used. 

[image]
[pin description.pdf](https://github.com/opentx/opentx/files/4639620/pin.description.pdf)
[USARTS.pdf](https://github.com/opentx/opentx/files/4639621/USARTS.pdf)

robustini, 2020-05-17:
It would be very interesting.

brainbubblersbest, 2020-05-17:
[image]
[image]

brainbubblersbest, 2020-09-25:
Another User makes this possible for Jumper16 and RadiomasterTX16. #7935 
So its no big thing, as i expected.

Also, there are Users out there who have this "issue"
#6737 #6938

kilrah, 2020-09-26:
Duplicate #7935

closed 2020-09-26 by kilrah