opentx/opentx #7611: EEPROM ALERT afert trying to access at SD CARD page

Armani88, 2020-05-17:
First of all, sorry for my poor english !

I tryed to update my X9E radio to the last version of OpenTX 2.3.7

So i downloaded:
- Companion 2.3 (companion-windows-2.3.7)
- The firmware 2.3.7 (opentx-x9e-lua-noheli-fr-2.3.7.bin)
- The SDcard file (sdcard-212x64-2.3V0026)

I uploaded correclty evrything in my radio and when i started again my radio, i have a request for eeprom conversion.
But there is another message about RTC batterie.
So, i changed the RTC battery CR1220 because i have the radio since about 5 years.

I repeat again all the process and now, it's the same (without the message about RTC battery), i have a request for eeprom conversion.
So, ok for convert eeprom and my start menu appear. 

Everythings seems to works perfectly.

But when i try to access to the page 2 (SD Card manager):
- Reboot of the screen
- the screen display "EEPROM ALERT, EEprom Data v217, button for continu"
- EEprom conversion
- The color top screen display a voltage at 0V for 1sec and then the good voltage
- The bottom screen back to the start menu but with an "Exclamation mark" on the RH top of the screen

So, i'm in a loop if i trying to get the page 2

**General information**
VERSION        Page 7/7:
- FW: opentx-x9e
- VERS: 2.3.7 (8b080683)
- DATE: 2020-03-30 16:24:26
- EEPR: 219

So it's raining today, and i'm unable to use my radio for a training on my favorite simulator ! 
The radio doesn't work.

Many thanks for your help !

3djc, 2020-05-17:
Could you try a 2.3.8 nightly ?

Armani88, 2020-05-17:
Hello 3djc and thanks for answering.
So, i just tryied the 2.3.8 nightly version (opentx-x9e-lua-noheli-fr-2.3.8N214) with companion 2.3.8N214 and the last SD card file (sdcard-212x64-2.3V0027.zip)
The problem for access to SD manager Page is fixed. I can display the page 2.
But:
- Everytime i start my radio i have EEPROM ALERT (EEprom Data v217) and need to convert it
- Everytime i unplug USB cable, also ALERT EEPROM and conversion

If i try to update the firmware and eeprom by the radio, (i copy/paste firmware and eeprom in my SD card before), it says "Directory is missing".
So, even if i have the last version of SD card file, i changed "EEPROM" for "EEPROMS" and same for FIRMWARE**S**
I perform the upgrade by the radio but same problem with conversion.

For information, it displayed: "Taranis BootLoader - 2.1.6
I thought the BootLoader folow the firmware version, no ?
Maybe, something i don't understand well !

3djc, 2020-05-17:
Sd card is not used for models on taranis. The bootloader hasn’t changed probably because you flash the radio while in bootloader mode.

If you are coming from 2.1.x, I would recommend going to 2.2.4 as an interim step, then 2.3.7

kilrah, 2020-05-17:
You should have conversion ONCE and that's it, unless you somehow then reload a 2.2 eeprom afterwards.

Armani88, 2020-05-17:
Yes i'm coming from 2.1.9.
So, I just went to 2.2.4 and evrything was perfect.
And i tryed again the 2.3.7 but it's the same issue. Crash at SD Card manager (page 2)

And tonight, i don't know why, impossible to dowload companion 2.3.6 for try if it's the same.
Download failed at the end !
But maybe it will be the same issue with 2.3.6 !

kilrah, 2020-05-17:
No point trying different companions, they will all give you the same firmware.

3djc, 2020-05-17:
Yes 2.3.7 does have this page2 issue on x9e. Use 2.3.8 nightly until we release 2.3.8 in not too much time

Armani88, 2020-05-19:
Thanks a lot 3djc and kilrah. No emergency ! Still waiting for my new FC. And then, i'll go to the release 2.3.8
Thanks for your support. I learned a lot !

closed 2020-05-19 by kilrah