opentx/opentx #7612: 2.3.8 - Poor brightness range of the LCD on X10S Express

robustini, 2020-05-18:
Unfortunately, even with the brightness set to the minimum, the LCD emits a fairly strong light.
From the video maybe you don't notice, but this problem does not exist on X12S and Jumper T16.
I don't know if it's the fault of the firmware or a hardware limitation of the LCD.
The minimum light intensity is such as to consume a lot of battery.
I'm sorry for the tilted video.

https://youtu.be/iH0rwoHh37k

 - OpenTX version 2.3.8 N215
 - Hardware Horus X10S Express

3djc, 2020-05-18:
I'm unusure for X10 Express, but X10 was a hardware limitation. Reducing bellow the limit we have put in place resulted in heavy screen flicker.

kilrah, 2020-05-18:
Would expect that part of the hardware to be the same, but someone might want to try.

closed 2021-02-13 by stale[bot]