opentx/opentx #7613: Lost of sensors if I save the model from PC to the radio using Companion Software

fdapport, 2020-05-18:
Hardware
Radio Taranis X9D+, XSR Receiver or R9Slim+, Betaflight F4 PRO V3 or F405Wing. PC with Win 10 PRO

Firmware
TX 2.3.7, RX XSR 2.1.0 or 190201 R9Slim+ , Ardupilot 4.0.5
Companion 2.3.7
I use the telemetry from RC controller trough Inverted PORT of the RX
I use as well the YAAPU script version 1.8.0

The problem:
When I read the models from companion software on the PC and I change same parameters in the models (also without any changes) and I save back I have some changes in the information on the sensors as the attached pictures. The parameter changed is the SENSORS ID, from 0 or 1 to 224 or 225, and Sensors source  from INTERNAL (Foto_3.jpg) to "External Module" (foto_11.jpg). Then I do not receive any values from the RC controller sensors but yaapu script works. To fix the issue I read again the sensors from the Telemetry Menu but I could have different sensors sequential number that could have issues in the special function menu of the radio, like Voice alarm. Yaapu script works also without to read again the sensors.

How to reproduce the problem:
On the radio
1) Check the sensors. The sensors ID=0,  1 or GPS=251 from what I received from the RC controller : Foto_2.jpg
2) Check the sensors  details #4 VFAS (battery voltage): Foto_3.jpg 

On the PC
3) Read the model from the radio: Foto_5.jpg
4) See the models that I have read from the radio: Foto_6.jpg
5) See the sensors detail in the telemetry Menu: Foto_7.jpg
5) Save Back the models with the following parameters: Foto_8.jpg and Foto_9.jpg

On the radio
6) Check the sensors, #4 VFAS NO VALUE and ID sensors is CHANGET from ID=0 or 2 to  ID=224 or 225!!!! : Foto_10.jpg
7) Check the sensors  details #4 VFAS (battery voltage) and you see the Sensors source is changed from Internal to External MODULE.: Foto_11.jpg. 
8) The alert voice alarms could be different after I read again the sensors.

Please let me know if you need addition information.

The pictures:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

yaapu, 2020-05-20:
I might add some context to this issue, looks like saving/restoring models with sensors created by LUA with the setTelemetryValue() fails.
Sensors created with the setTelemetryValue() are discovered with an ID=0 but if saved and restored by companion are recreated with a different ID, 224 in the above screenshots

bsongis, 2020-05-31:
For the sensors IDs we use the 3 high bits to store the module (1 bit) and Rx ID (2bits). And when those 3 bits are set to it means the sensor is found on S.PORT line (we don't know which module and there is no receiver ID)

So here 224 (11100000) means a sensor ID=0 on S.PORT line

You are on Taranis X9D+ which has no ACCESS, so you should only have S.PORT sensors, all beginning with the 111

bsongis, 2020-06-01:
I am afraid that my fix will break the RSSI when RSSI will come from the R9M (ACCESS) and ISRM (ACCESS).

fdapport, 2020-06-01:
I use the FC controller Betaflight F4 PRO V3 and arupilot Ardupilot 4.0.5, FrSky SPort Passthrough (SERIALx_PROTOCOL=10) and RX/TX  accst protocol and LUA YAAPU script... what about Sourse=INTERNAL that it changed from Companion software when I save the model in External module (see pictures) ?

closed 2020-06-04 by 3djc

fdapport, 2020-06-04:
Do you have close it because it will be fix in the new release??

3djc, 2020-06-04:
yes

fdapport, 2020-06-04:
Great,, Thanks, Franco

commit a468005a (2020-06-04): 2nd attempt to fix #7613 (#7670)