opentx/opentx #7614: rotary encoder handling on Horus X10 with INTERNALACCESS

ParkerEde, 2020-05-18:
Hi,
my old X10S with iXJT works fine with the rotary encoder on OTX 2.3.7.
After updating ISRM Upgrade module and OTX to 2.3.7 INTERNALACCESS the rotary encoder makes sometimes some jumps to the wrong direction.
I've double checked: 
Go Back to iXJT and OTX 2.3.7 without INTERNALACCESS and rotary encoder works fine. 
Now X10S is running ISRM and OTX 2.3.7 with INTERNALACCESS again and the rotary encoder makes sometimes some jumps to the wrong direction if I'm scrolling through the menues.

Is it possible, that the changes from 2.3.0 to 2.3.1 "Improve rotary encoder handling on all radios" are not in the branch of INTERNALACCESS?

Kind regards
Ralf

3djc, 2020-06-14:
Also mentionned here : https://www.rcgroups.com/forums/showpost.php?p=44761463&postcount=645

ParkerEde, 2020-12-11:
just a post to "unstale"

bsongis, 2020-12-11:
Still not good even in 2.3.10?

ParkerEde, 2020-12-11:
no, sorry