opentx/opentx #7620: 2.3.7 ACCESS telemetry lost alarms although RSSI is updated

Allerhopp, 2020-05-22:
X9D+ 2019, OTX 2.3.7  G-RX8 ACCESS 202000519 beta firmware

Some "telemetry lost" alarms were marked in the logfile with a SF switch action. But no data freeze could be found when this happened. RSSI is expected to be frozen for 1s but apparently it was not. Is there a chance that this "telemetry lost" alarms are false alarms? ACCESS is prepared to handle more than one RSSI. Eventually there is a context? With ACCST the frozen RSSI was always visible when telemetry lost was triggered.
https://www.rcgroups.com/forums/showpost.php?p=44574403&postcount=99
Logfile and some details:
[INDOLORE_F5J-2020-05-21.zip](https://github.com/opentx/opentx/files/4667279/INDOLORE_F5J-2020-05-21.zip)
[Detail.zip](https://github.com/opentx/opentx/files/4667282/Detail.zip)

Djodjotje, 2020-06-28:
same question here..... with the Xlite Pro 

XLite pro 2.1.0, OTX 2.3.9 G-RX8 ACCESS 202000519 beta firmware

mpjf01, 2020-06-29:
There's an issue with some G-RX8 receivers and loss of telemetry. See https://github.com/FrSkyRC/Firmware-Test/issues/60 and the RCGroups post linked there.

closed 2020-06-29 by kilrah