opentx/opentx #7621: 2.3.7 Jumper T12 Pro - GPS sensor not showing valid coordinates

peteoz, 2020-05-22:
On my Jumper T12 Pro display screen I only get a 6 digit number where the value of GPS coordinates is supposed to show up (the "GPS" sensor)

on Telemetry page under sensor "GPS" I also get the same 6-digit number followed by 3 letters "low"

I don't understand how is this of any use or how is one supposed to convert this to useable GPS coordinate format to locate a lost drone, this is not a valid GPS format (not even the new single string Google Plus Code)

is this a bug ?

3djc, 2020-05-22:
What gps sensor ? What type of receiver and telemetry ?

peteoz, 2020-05-22:
> What gps sensor ? What type of receiver and telemetry ?

I'm using HGLRC Ublox M8N GPS, TBS Crossfire Rx and running Betaflight 4.1.5

on my Betaflight OSD coordinates show correctly as 2 separate values (longitude and latitude).
also when running exactly same setup on Taranis radio everything works as expected and the GPS sensor displays 2 separate values (Lon + Lat)

only on Jumper T12 Pro I get this 6-digit number followed by "low".

My concern is that the Googles DVR not always records, I found occasions (mostly during crashes) that the Fatshark Goggles DVR fails to record and I can only rely on my radio.

peteoz, 2020-05-22:
here are few photos of what I'm getting on the screens and on Telemetry page.
(in my "Radio Setup" for "GPS Coords" settings it doesn't matter whether I choose "DMS" or "NMEA" format, the result is same)

[image]

[image]

[image]

kilrah, 2020-05-22:
Delete and rediscover the sensor

peteoz, 2020-05-22:
that worked. thanks !

closed 2020-05-22 by kilrah