opentx/opentx #7623: Flight mode transition bug

pmackenzie, 2020-05-23:
Another RCG user posted a problem he was having.
I looked at his OTX, and it seems like a bug to me in how flight mode transitions are done when a mixer channel is over +/-200.  The value changes sign and is offset by the difference from 200 during the transition.

Simple OTX attached, only one mixer line and one flight mode.  Channel value changes sign during flight mode transition, even though flight mode change should not affect its value at all.

pmackenzie, 2020-05-23:
I don't see the attachment, above so here is a link to it on RCG
https://www.rcgroups.com/forums/showpost.php?p=44586343&postcount=2521

mpjf01, 2020-05-23:
Seen this before .......but where? Maybe closed issue?

dgatf, 2020-05-31:
@bsongis, weights over 200 is causing _sum_chans512_ overflows at the mixer when changing the flight mode. If we want to catch that, not sure which is the best option:

- Limit _chans_ value
- Decrease GV_RANGE_WEIGHT to 200. Is there a reason to have max weight up to 500? But still gvar could go up to 1024
- Decrease MAX_ACT from ffff to 3fff. This increases the fade in/out duration max error from 1 to 4.2s (if I understood it correctly)
- Change sum_chans512 to int64_t. Increases ram usage by 1kb and increases Tmix max from 0.45ms to 0.50ms (I guess this is max mixer processing time)

3djc, 2020-06-14:
Too be fixed in 2.3.10

PR #7690 (2020-06-07): Add dinamic bit shift for flight modes fading - Fixes #7623
Fixes #7623 

Add dinamic bit shift to flight modes fading

Advantages:
- Chans value never overflows the mixer when fading flight modes
- Increases accuracy for low chans values
closed 2020-06-17 by bsongis