opentx/opentx #7625: Error Crossfire Outmap

toper74, 2020-05-23:
Hi,
i just update my Qx7 to Otx 2.3.7 and when i go to Tools > Crossfire config > Nano RX, i can scroll to the first 7 options (General, Mode, Telemetry, Rf profile, Failsafe mode, RC by Mavlink and Factory reset), but when i try to go to "Outmaps", i have this error message: "Error syntax script bad argument #3 to 'draw'......'
Can't see the full message.
I have this issue on all of my setups (drone)

3djc, 2020-05-23:
No issue here : 
[image]

You many need to try an older crsf version, some run out of memory

toper74, 2020-05-23:
An older CRSF version, and so an older versopn of OTX (or not) ?

3djc, 2020-05-23:
This was shot with 2.3.7 as you mentionned that version, with CRSF 3.28, works fine here

toper74, 2020-05-23:
So i have a memory issue on my Qx7 ?

3djc, 2020-05-23:
No probably not.
You need to make sure that:
- crsf speed is set to 115k in radio/hardware
- you have the right SD content, especialy that you do not have some old scripts.

toper74, 2020-05-23:
Sorry but can't find crsf speed option...
Is it on the crossfire config menu or Taranis menu ?

3djc, 2020-05-23:
Taranis, Radio, then hardware (6/7), then line called max bauds

toper74, 2020-05-23:
Found. But same issue...

3djc, 2020-05-23:
Please make sure you only have the content of our pack on your sd (at least for testing)

toper74, 2020-05-23:
After upgrading, i format SD Card and put the "sdcard-128x64-2.3V0026.zip " from here https://downloads.open-tx.org/2.3/release/sdcard/opentx-x7/

3djc, 2020-05-23:
Yes I’m doing exactly that

toper74, 2020-05-23:
And that's what I did... So?...

toper74, 2020-05-23:
Oh my god, i'm so dumb..... was trying without antenna on Qx7....
I'm so sorry !...
Thanks for helping !!

3djc, 2020-05-23:
You are going to fry your crossfire doing that ..

closed 2020-05-23 by 3djc

kubinets-ru, 2022-04-28:
I have the same problem, I have checked and updated everything, all antennas are connected but nothing helps. Could you help anybody help me?
[image]