opentx/opentx #7629: Taranis x9d + 2019 wont turn on after flashing under some options

suhast222, 2020-05-24:
Hello I have a problem with my taranis, whenever I try to upgrade opentx firmware (I have tried 2.37 and 2.38) the radio wont boot under some complation options.
For example It happens if I don’t check ppmus and nogvars. If I uncheck that options the radio works fine.

3djc, 2020-05-24:
What is the full name of the non working firmware?

suhast222, 2020-05-24:
A lot of them won´t work for them. For example:
opentx-x9d+2019-lua-noheli-sqt5font-en-2.3.7.bin does not boot

This one is one of the few that works fine:
opentx-x9d+2019-faichoice-flexr9m-lua-nogvars-noheli-nooverridech-ppmus-sqt5font-en-2.3.7.bin

3djc, 2020-05-24:
Have you put a firmware on the sd card , start in bootloader and flash from there ?

suhast222, 2020-05-24:
Bingo! That did the trick.
What is the problem then? The companion or my PC? I have tried in my desktop and my laptop, both windows 10 with the same results.

3djc, 2020-05-24:
Most likely your usb cable

closed 2020-05-24 by 3djc