opentx/opentx #7631: No telemetry input in Free Link App from built in blutooth with X10s or X-Lite/S

StefanKellerAC, 2020-05-25:
I try to catch telemetry output from built in Bluetooth with a HC-08 BLE 4.0 Module.

- With OTX 2.3.7 on the X10s ( also on a X-Lite/S ) connection can be  established to the following partners, a G-RX8 is connected, but:
	-**no** Data is received with the Free Link App 
       -Data **can** be received in BLETerm on Android, at 0000FFF6-0000-1000-8000-00805F9B34FB
	-**no** Data is received with a connected HC-08 BLE 4.0 module ( set to the characteristic with AT+TUUID=FFF6 )

- downgrading my Horus X10s to OTX 2.2.4 gives Data from the connected G-RX8 
	-in the Free Link App 
	-with the HC-08 at 0xFFF6
	-with BLETerm at 0000FFF6-0000-1000-8000-00805F9B34FB

( Trainer connection between those transmitters works in 2.3.7 )

StefanKellerAC, 2020-05-26:
I changed nothing :-) but Data transfer with the HC-08 Module and x10 and 2.3.7 and X-Lite work now,
also with an HM-10 Module  ( AT+CHAR0xFFF6 )
but Free Link app still receives **no** Data...

brainbubblersbest, 2020-06-20:
+1 on this. Bind a hm10 to internal bt via at command. Need S. port out for a mfd crossbow Mini. Bt module with inverter connencted to the tracker. 
On 2.3.2 this have worked flawless.
Bt module blinking led has shown active data transmission on the hm-10 and hm-18 bound to x12s.
On 2.3.7 here is definitely no data active indicated by total dead data led. 
Has this maybe to Do with that bt fifo issue for crossfire, wich was merged into this Version?

brainbubblersbest, 2020-07-11:
> I changed nothing :-) but Data transfer with the HC-08 Module and x10 and 2.3.7 and X-Lite work now,
> also with an HM-10 Module ( AT+CHAR0xFFF6 )
> but Free Link app still receives **no** Data...

Can you explain how you  change the char to 0xFFF6?
Tried everything. Like it's hardcoded to FFE1. 
Do you use a specific Firmware on that module?

StefanKellerAC, 2020-07-11:
no, I did nothing special with my module, just typing in AT mode:
AT+CHAR0xFFF6

http://fab.cba.mit.edu/classes/863.15/doc/tutorials/programming/bluetooth/bluetooth40_en.pdf

brainbubblersbest, 2020-07-11:
I typed the same. AT+Char? Gave me just the old value. Ind the x12s Flash
download archive is a blemodule.bin is it possible to Look into that
Firmware?

brainbubblersbest, 2020-07-12:
Found this in another project. 
They figure out that some crc checks changed too. Anyway, the external s.port connector is not Hit by that. 

https://github.com/raul-ortega/u360gts/issues/91

Maybe, when it's not opentx related. 
Anyone can say it's frsky related then? 
(but the behavior changes if use a older otx Version)

StefanKellerAC, 2020-07-15:
the Free Link App is still not working on X10 with 2.3.9

arf-it, 2020-07-17:
> I changed nothing :-) but Data transfer with the HC-08 Module and x10 and 2.3.7 and X-Lite work now,
> also with an HM-10 Module ( AT+CHAR0xFFF6 )
> but Free Link app still receives **no** Data...

Hi @StefanKellerAC may I ask what the  AT+CHAR0xFFF6 is doing and why is needed?

Thanks

StefanKellerAC, 2020-07-17:
It sets the characteristic ("address") where the data is received.

closed 2021-06-04 by stale[bot]