opentx/opentx #7633: Labeling Simulator GUI

bigtwin, 2020-05-25:
When working with the simulator in order to check my configuration, it is always kind of tricky to actually find out what is going on:

- If I move a switch upwards on the screen - is that actually towards me on my transmitter? Or right the other way?
- If I want to influence a channel and its output bar moves downwards - is that actually a flap moving downwards or upwards

In order to make this more convenient, it would great if I can somehow attach a label to the controls of the transmitter GUI of the simulator. 
Everytime I would know e.g. that moving the GUI switch down, means to actually move the switch on the real transmitter towards me.
Also, it would be great to label e.g. channel 5 in the outputs window as _AilRight up_ and _AilRight down_. If I am now working with the configuration and I observe that channel 5 output increases by 13%, I know right away that this means right aileron moves downwards.

These labels can be stored with the model and I would always be aware how my model is build.

bigtwin, 2020-05-28:
Both issues are not related! The only thing they have in common is the fact that it is about documentation features in Companion! 
Please re-open #7645 or include its topic to this issue here.

BTW - I have discussed both topics with other OpenTX/Companion users and they also have the same problem and would appreciate some documentation feature as well.

Thx for taking care of OpenTX and Companion!

Mateyhv, 2020-06-01:
moving the switches is quite intuitive. Front switches act the same way on TX and on GUI. Top switches you should look them from the top so there is no towards me or away from me but again up and down. Labels on channels would be nice to see though.

What I would like to see is realistically positioned switches, each time I have to stop and think about the switch letter, otherwise I always flick the wrong switch in the simu. Would be much more easy if switches had the same distribution as in the transmitter.

closed 2021-03-20 by stale[bot]

timtuxworth, 2021-06-18:
Just want to add my two cents that this would be an incredibly valuable feature.