opentx/opentx #7644: TBS nightly build, D16 RC Command, frame rate issue.

0crap, 2020-05-28:
**Sorry if this does not belong here, I first opened a ticket @ TBS but they told me they are not able to solve the issue and pointed me to the OpenTX devs.
So I open a new issue here, if this is not correct please feel free to close it! Thank you.**

So, I'm running the TBS nightly OpenTX version on my FrSky Taranis QX7S to enjoy CRSF Shot.
http://team-blacksheep.freshdesk.com/support/solutions/articles/4000153333-crsf-shot-open-tx
This works fine until I switch to D16 mode and fly my other FrSky equipped quads.
The problem is my RC Command traces look horrible and always have 18mS interval. 
[image]
In debug mode the RAW RC Command input also shows 18mS intervals.
[image]
This gives problems if you run a lot of FeedForward and causes unwanted motor jerking.

If I switch to OpenTX 2.3.7 and not change anything else, all problems are gone.
(But now no CRSF Shot of course.)
Always nice 9mS RC Command intervals as expected.
[image]

Maybe this will not be fixed until OpenTX 2.4 is out.
It's just I want to make the devs aware of this.

If anyone else using the TBS nightly OpenTX build can confirm my findings would be great.
Thanks for your time.

3djc, 2020-05-28:
TBS build are are a snapshot of 2.3.5 wich does not include any later changes. This is per design, and TBS where informed when this special firmware was created.  The plan is to have CRSFshot part of mainstream OpenTX in 2.4

closed 2020-05-28 by 3djc

0crap, 2020-05-28:
I fully understand @3djc .
Looking forward to OpenTX 2.4

It would be nice if TBS gives this info on their website as a downside of using their nightly build.
I asked them to do so, just to keep users informed correctly.

Thanks again for your time.