opentx/opentx #7645: Provide description field for Logical Switches and Special Functions

bigtwin, 2020-05-28:
It would be helpfull to have a description or let say comment field for a logical switch or a special function. Only in Companion an optional field which allows me to describe what for I have made this line and was it intenion was.

Every time I want to change a model config after I havent touched it for a while, I have difficultied to decrypt my thought son LS and SF. Would be great if I can label them somehow.

3djc, 2020-05-28:
Duplicate  #7633 ?

closed 2020-05-28 by 3djc

bigtwin, 2020-05-28:
No! Not at all a dublicate!
This is about a comment field for Logial Switches and Special Functions, while the other issue is about the Simulator GUI and some labels to the switches and output bars.

I kindly ask to reopen this issue or add this topic to the still open issue

kilrah, 2020-05-28:
Neither are not possible without a complete redesign of the storage and we've got that use case in mind whenever that happens one day.

bigtwin, 2020-05-29:
Thx for the statement and considering it someday!
But just in order to prevent any misunderstanding again - I am talking only about Companion and NOT the OpenTX itself.

kilrah, 2020-05-29:
Yes but currently any data visible in one must be in both, so any comments for companion would have to be stored in the radio which doesn't have enough memory for that.

ridietz, 2020-10-13:
I would find this very helpful also.  I have been putting a blank line between Logic Switches and Special Functions so I can make comments on hard copies for future reference.

ridietz, 2021-03-05:
For the majority of users, the small amount of memory in the radio would not be a problem if comments could be used in logical switches and special functions.  Other users could avoid these comments when memory capacity was a problem.  Thanks for your consideration 😷

bigtwin, 2021-03-06:
I also want to point out again, that such a comment or description feature would not have to be assigned to the actuall program or configuration code. Even an additional tabulator side in _Companion_ for the model would  be a big improvement. If I would be able to store some additional comments along to each model - just and only in companion, not in OTX and just as free text and nothing special, it would be a big and great help.
I am thinking of just another tab, may after _Special Functions_ and _Telemetry_ - would do the job.

Thx again for considering to reopen the issue.

lshems, 2021-03-06:
Anybody interested in this feature, please contact me on RC groups, pm l Shems.

I'll try to figure out a lua workaround based on some spec gathering.

ridietz, 2021-03-08:
Thanks, I am interested.  I do not know how to contact you at RCGroups.  I am ridietz on RCGroups.

Dick Dietz, 317-560-0219