opentx/opentx #7646: tRandom, non intentional "receiver still connected" warnings while flying

knoopx, 2020-05-29:
https://youtu.be/2KBTGwhkIzQ

Happening on jumper t12/2.3.7

3djc, 2020-05-29:
What happens if you have no receiver and set power off to instant, does it shutdown?

knoopx, 2020-05-29:
Haven't tried it yet, gonna try tomorrow. My guess is nothing will happen as I get no visual notification to confirm power off, just audio/haptic feedback.

kilrah, 2020-05-30:
The power circuitry might see a short blip on the button signal, too short to get the LCD to respond yet..

What protocol are you using, on which module?

knoopx, 2020-05-30:
Frsky d16 on internal multiprotocol  v1.3.1.1

knoopx, 2020-05-30:
Ok, just tried without the power off delay, the radio does not power off by itself but I can still hear the audio/haptic feedback.

kilrah, 2020-05-30:
He asked to try without a receiver powered, to see if the radio would just randomly shut off.

I've just left my T12 Pro on N223 with MPM 1.3.1.4 on FrSky D16 bound to an S6R for an hour with no spurious message, so it's likely a hardware issue somewhere.

A short press of the power button indeed doesn't show on screen.

knoopx, 2020-05-30:
Just tried N223 and happened too. Only happens when connected to a receiver. The radio never powers off by itself regardless if connected or not.

@CurryKitten also said:

> I've seen it happen as well, but only when I've been in very close to the RX.  Making me think it was a symptom of the RX getting flooded (where you'd normally just get a telemetry lost)  Have either of you guys had it out on the field where you are proper distance away ?

knoopx, 2020-05-30:
BTW, whenever I turn on the internal module the speaker makes noise. I wonder if this issue might be related to interference? @kilrah does yours make noise as this one? https://www.youtube.com/watch?v=2dj8QXKpPRA did you MOD it?

3djc, 2020-05-30:
Interference noise is terrible on t12, really bad. But only hurt your hears

knoopx, 2020-06-01:
Today I flashed 2.3.7 and ran through 10 packs with `disableRssiPoweroffAlarm` off and `pwrOffSpeed` = 0 and the radio never powered itself down.

Some users and myself also get `AU_MODEL_STILL_POWERED` just after powering off the receiver.

gregherrell, 2020-06-14:
New T12 Pro using FrSky D8.  Tx came with 2.3.6 installed.  Internal module was 1.3.0.53.
I would get random "Receiver is still connected" .wav messages along with a Tx vibration.  Accompanied at times by Rx Loss and the whoop falling out of the sky. The radio never powered down.

Updated to 2.3.8 and 1.3.19.  Also did the recommended Frequency Tuning recommended in the MM documentation.  I still get random "Receiver is still connected" .wav messages with the vibration.  In fact, I can hook the quad up to BetaFlight, leave the Tx on and it will happen sporadically without any input.  After several packs today I did not get Rx loss.  However, while flying I now get intermittent "Telemetry Lost, Telemetry Recovered" .wav messages.  However, I have yet to experience Rx Loss.  Rss strength does not budge during these alert events.

CurryKitten, 2020-06-19:
I just updated to 2.3.9 and put the sd card back in (after having it removed for the last few weeks because of #7505 I noticed several "Receiver still connected" warnings whilst just having a quad on the bench doing some stuff in Betaflight and when connected to a quad with no inputs.  I was using D8 protocol.

If this is down to a hardware issue which means the firmware perceives that the power button has been pressed, is a workaround to wait for a button press of, for example, 1 second before playing the alarm?  Since you still have another few seconds of pressing the button before getting the alert box asking you to confirm that you want to shut the radio down.

3djc, 2020-06-19:
Well surely if it is down to hardware, the solution is fix hardware ?

CurryKitten, 2020-06-19:
> Well surely if it is down to hardware, the solution is fix hardware ?

Oh sure, but multiple "ifs" if it's hardware, if it's fixable.  I was thinking more of if it's a limitation of the hardware.  But it's all conjecture at this point, so I'll be quiet.  Anything I can do to help determine the issue ?

knoopx, 2020-06-19:
@CurryKitten I implemented the software "patch", but pretty sure devs will be reluctant to merge upstream.

longxk, 2020-06-25:
the "receiver still connected" warning appears randomly on my original t12 too.
but only after I installed the internal multi-protocol module and update the firmware.

kf4mot, 2020-06-25:
If noise is suspected I would try adding a low value cap across the power button to see if it improves. Something like 100 to 1000 pF.

OMGChase, 2020-08-03:
Hello. Brand new T12 Pro user her. Having the same thing. Haven't been able to fly yet because of weather so I can't testify to rx loss yet. Will update once I get to it.

Gomer05, 2020-09-16:
Jumper T12 Pro
OpenTx 2.3.9 
MPM 1.3.1.49 and 1.3.1.59 
I also have the issue where "receiver still connected" comes up randomly, have rebound and "tuned" the frequency as well Thanks @CurryKitten

frizzle101101, 2020-10-05:
my new t12 has this issue too, i have also updateded to newest mpm and opentx firmware

SergeyTomp, 2021-03-16:
Jumper T12 Pro - brand new, bought at March, 02, 2021 
OpenTx 2.3.11 + MPM 1.3.1.59
"Tuned" frequency.
Issue is present still.

knoopx, 2021-03-16:
I have a fix [here](https://github.com/knoopx/opentx/commit/835252fc7ae33c06cb66bfbf00a0e3a27b37b247). I don't think its a hardware issue at all.

3djc, 2021-03-16:
Well your fix is within if (pwrPressed()), which proves there is an hardware issue since you should not be getting pwrPressed() if you are not actually pressing power

knoopx, 2021-03-16:
@3djc or that gpio/watchdog is not properly setup

frizzle101101, 2021-03-16:
I believe this is also probally hardware issue no other tx has it...maybe
adding a cap or something else, but a 0.1uf culdnt fix it sofar, may have
reduced the frequency of it happening but culdnt say forsure, dont have
good enough testing equipment to see the issue maybe somthing else to try,
it is strange

SergeyTomp, 2021-03-19:
> I have a workaround [here](https://github.com/knoopx/opentx/commit/835252fc7ae33c06cb66bfbf00a0e3a27b37b247). I don't think it's a hardware issue at all.

This workaround will work until new release come. It will be overwritten during flashing and appear again if this bug not fixed still.
The more simple way is to rename modelpwr.wav or delete it at all from sd-card. You will never hear this announcement after this. 
This is still safe because power off confirmation menu is still shown before receiver could be switched off if you really press and hold the power button.

closed 2022-04-17 by stale[bot]