opentx/opentx #7648: Severalissues related to model.getSensor(sensor)

rdeanchurch, 2020-05-30:
The documentation (2.3.0) for model.getSensor(sensor) is essentially a copy of  get logical switch in relation to return values .
The correct documentation should be closer to table telemetry sensor data:
		  Custom Sensor
 			 name,  (string) A2
 			 id, 	  (number) 61699
 			 instance, (number)  250
			 unit,  (number) 1
 			 type,  (number) 0
 			 prec,  (number) 1
			Example 
 			 i=type, v=0
 			 i=unit, v=2
 			 i=name, v=Curr
 			 i=id, v=512
 			 i=instance, v=226
 			 i=prec, v=1
		 Calculated Sensor
			 name,	(string) mAhr
			 prec,	(number) 0
			 formula,(number) 7
			 unit,	(number) 14
			 type,	(number)  1
			Example 
			 i=type, v=1
 			 i=unit, v=14
 			 i=name, v=mAhr
 			 i=formula, v=7
 			 i=prec, v=0

2. The following Telemetry Sensor parameters are not available and would be useful:
			  ratio,	(number)
			  offset,(number)
			  autoOffset, (boolean)
			  filter,     (boolean)
			  positive,   (boolean) 
			  persistent,  (boolean) 
			  log,  (boolean)

3. An empty sensor still returns a table, and should probably return a nil.  Presently custom function values are returned with all zeros for parameters.
  i=type, v=0
  i=unit, v=0
 i=name, v=
 i=id, v=0
 i=instance, v=0
 i=prec, v=0 

4 It would be highly desirable to have a companion function,
 model.setSensor(sensor, parameterTable)

All tested only in 2.3.7 Companion.

rdeanchurch, 2020-06-04:
5. the values returned for instance need to be ANDed with 0x1f then have one added to look the same as they do on the Telemetry page.

strgaltdel, 2020-09-22:
Just ran into same situation,

a "model.setSensor()" function would be nice,

DaySpeed955i, 2020-12-30:
I am now using ACCESS and model.getSensor() would be very useful with "Source" in the return table.
It would then be possible to know which receiver is being used to return telemetry when multiple receivers are used.

Example under ACCESS
Source    SR10-SP1

Thanks

closed 2021-10-02 by stale[bot]