opentx/opentx #7649: Para bluetooth trainer not working when X-Lite Pro is master

lthole, 2020-05-30:
Similar to X10S Express issue #7094.

If I set up my X-Lite Pro as BT master, and my X9D+2019 as slave, the stick values next to Cal on the Trainer screen do not move, i.e. the slave does not come through.

Test No 1:
----------
In the X9D+ SE 2019, I have the following:
a) Bog standard model, 4channels AETR.
b) In Hardware, Bluetooth is set for Trainer.
c) In Model Page, Bluetooth is set as Slave/Bluetooth.

In the X-Lite Pro, I have the following:
a) Model is set up for Slow Stick, channel order AETR.
b) In Hardware, Bluetooth is set for Trainer.
b) In Model Page, Bluetooth is set as Master/BT, then hit discover, and select the BT code for the slave. 'Discover' changes to 'Clear'. 
c) In Hardware page:
Local addr 806FB0FE82F8
Dist addr 806FB0963467
d) In Special Functions, switch SD down enables Trainer function in Sticks mode, SF box checked.

1. Power up model on X-Lite Pro, check servos and ESC work.
2. Switch SD down.
3. Slave does not control model. Look in Trainer page,  the stick values next to Cal on the Trainer screen do not move.
4. Check that the both addresses still show in Hardware page.

Test No 2:
----------
Swap all above steps between X-Lite Pro and X9D+ SE 2019.
The Trainer works as expected.

lthole, 2020-05-30:
Both radios are on OpenTX 2.3.7.
Probably not relevant to this issue:
X-Lite Pro: ISRM-PRO 1.1.0/2.1.0 FCC
X9D+ SE 2019: ISRM 1.1.0/2.1.0 FCC

bsongis, 2020-05-30:
Are you sure you have flashed to BT module on both radios to the latest version?

bsongis, 2020-05-30:
I just checked here, it works perfectly

kilrah, 2020-05-30:
I don't think FrSky have published any BT firmware...

lthole, 2020-05-31:
Further information:
With the X-Lite Pro as master, and X9D+ SE 2019 as slave:
a) As expected, the X-Lite shows both BT codes in the Hardware page:
Local addr 806FB0FE82F8
Dist addr 806FB0963467  

However:
b) The X9D only shows 
Local addr 806FB0963467
Dist addr ---
When testing the other way around, the slave was also showing both BT codes.

c) When I enable the Trainer function with switch SD down, the X-Lite still controls the model. I was expecting '0' to be used as slave values since the link is not functioning.

b14ckyy, 2021-03-12:
Hey guys.
I have the exact same problem. 100% identical to what @lthole describes. In my case it is a X10S BT module installed in a Radiomaster TX16S. 
Para trainer works when the X-Lite is master. If I swap over the settings to make TX16S master, the trainer switch on the TX16S is just ignored and the X-Lite slave is not given control. BT connection is stable and both radios show the distance addresses correct.  

Both Radios are on 2.3.11 btw.

b14ckyy, 2021-09-11:
Is there any news about this? Not tested again.

brainbubblersbest, 2021-09-11:
No. But you can solder the Para bt to your TX16S instead of the old Horus BT.