opentx/opentx #7650: Trainer page should adapt to 'Channels' vs 'Sticks' Trainer modes

lthole, 2020-05-30:
With reference to the new feature introduced in OpenTX 2.3.3 'Trainer function can now replace the channels values instead of the sticks positions':

The Trainer page is now misleading if the 'Channels' mode has been selected in the Trainer SF.

The left hand column always shows the master's side with stick symbols, which is fine when using the 'Sticks' mode. However, when 'Channels' mode is selected, the master's side should change to CH1 - CH4.

3djc, 2020-05-30:
This could become very complex as you can set a SF that does stick, another channel ... and watching with neither actived

lthole, 2020-05-30:
Or maybe make the screen more generic by not having a stick symbol?

kilrah, 2020-05-31:
Trainer Channels bypasses the ENTIRE master radio, so the page is irrelevant.
It just takes the slave's signal and puts it out straight to RF.

closed 2020-05-31 by kilrah