opentx/opentx #7659: Qx7 Access + r9m (Old, not 2019) flex = no link

d2f2, 2020-06-01:
R9m pre2019 cant work with 2.3.7 Qx7 Access. 

Firmware for Qx7 Access includes flex. 
Firmware for pre2019 is flex. 
Firmware for r9mm is flex. 

R9m works fine with old qx7 accst, but not the Qx7 Access one. 
Link is absent, red light shown at r9mm. 
R9m set as r9m, not r9m access in Qx7 Access. 

**To Reproduce**
Steps to reproduce the behavior:
1. Flash Qx7 with 2.3.7 with flex, luac, font. 
2. Flash r9m as flex (flashing r9m works fine with Qx7 Access) 
3. Set external module to r9m (not r9m Access). Unable to bind receiver, Unable to establish a link. 

**Expected behavior**
Expected link and binding works. 
The same module in Qx7 accst works fine woyh receiver at flex 8 6 8.

**General information**
 - OpenTX version 2.3.7 flex luac gont noheli
 - Hardware Qx7 Access, r9m before 2019

kilrah, 2020-06-10:
None of us has a QX7 Access, but we tested on X9D+2019 which should be the same functionally and had no problem.

bsongis, 2020-06-10:
Will be tested by FrSky by tomorrow.

d2f2, 2020-06-10:
I have tested today with r9m 2019 flashed to newest accst flex firmware.

R9m 2019 works fine in accst Mode with Qx7 Access. 
R9m non 2019 not working in accst Mode with Qx7 Access.

bsongis, 2020-06-10:
And if it is not FLEX ?

d2f2, 2020-06-10:
I am interested only for flex.

bsongis, 2020-06-10:
I can suppose so, I ask this just to be able to track this bug

kilrah, 2020-06-10:
Flex or not should not change anything

bsongis, 2020-06-11:
X7Access + old R9M + Flex firmware tested at FrSky. Bind worked. I have seen a video taken yesterday. We need more input on this. Is the FCC / EU R9M firmware working right? And XJT?

Do you have a logic analyser?

kilrah, 2020-06-11:
Does another module work on the QX7 Access? Module slot might have an issue.

d2f2, 2020-06-11:
> X7Access + old R9M + Flex firmware tested at FrSky. Bind worked. I have seen a video taken yesterday. We need more input on this. Is the FCC / EU R9M firmware working right? And XJT?
> 
> Do you have a logic analyser?

I will try to make logic analyzer. Also i ll try to make eu /fcc set of r9m and receiver, have to test before in x9d+ and Old Qx7. 

> Does another module work on the QX7 Access? Module slot might have an issue.

R9m 2019 works fine with accst firmware. 
Multi protocol external jr Jumper module from t16 Also works fine.

bsongis, 2020-06-11:
From opentx door, R9M-pre-2019 and R9M 2019 with an ACCST firmware are exactly the same

kilrah, 2020-06-11:
Can you confirm you have the internal module off when trying?
Does the multi protocol module correctly show its version in the menus?

d2f2, 2020-06-11:
> Can you confirm you have the internal module off when trying?

Confirm. Internal module is off. 

> Does the multi protocol module correctly show its version in the menus?

Yes. It shows corect info. 

> Do you have telemetry correctly with the 2019 module?

Rssi and receiver voltage is corect.
Taranis shows warning about power off when receiver is still on, warns about lost telemetry when i turn off receiver, says about telemetry recovered when receiver is back on. 

I will try later full telemetry from bf. Waiting for friend FC. 

> On the pre2019 module, what does the LED do on the module? when it is powered up with no receiver present?

Red light. 
And on pre 2019 is always Red light.

kilrah, 2020-06-11:
OK, and the red light doesn't turn off after a few seconds, right?

d2f2, 2020-06-11:
Red light on r9m 2019 when receiver is off is glowing, doesn't turn off after a few second. Tested few minuts.

d2f2, 2020-06-23:
Unsolved. Changed to r9m 2019.

closed 2021-03-21 by stale[bot]