opentx/opentx #7662: Not able to register to UID 1

Quantumx3, 2020-06-02:
Hello,
I tried out a multiple ACCESS receiver setup with my Horus X10S Express yesterday. Therefore I wanted to register a G-RX8 to receiver number 4 and to UID 0. That worked fine. Then I proceeded with a RX4R on the same receiver number, but tried to register to UID1. Everytime I chose UID 1 and pressed enter, it either switched to 0 or 2. That  issue occures in any model and with any receiver number I tried. The only working UIDs seem to be 0 and 2. Is that a bug or got the number of UIDs limited from 3 to 2?

Is there a option to receive the RSSI value of multiple Receivers simultaniously, like advertised by FrSky with TrioControl?

Thanks for making the best transmitter firmware! I look forward for help.

3djc, 2020-06-02:
No you cannot do that trio control as yet to come to reality, only one receciver sends telemetry

Quantumx3, 2020-06-02:
Thanks for that information about TrioControl. I am looking forward to it becoming reality.
What is about registering to UID 1?

bsongis, 2020-06-02:
Register to UID=1 should work!

Quantumx3, 2020-06-02:
Sadly enough it does not work. I made a quick video that shows that issue and uploaded it to youtube: [https://www.youtube.com/watch?v=zTic6AD1KnE](url)

bsongis, 2020-06-02:
I will test it by myself later today

RetoD7, 2020-06-03:
Basically it is working. Just the displayed value is one step behind during changes. The display is not updated or updated before change.
UID = 0 -> [Enter] -> [Up] -> still "0" -> [RTN] -> UID=1 -> "all okay"

Quantumx3, 2020-06-03:
Oh man, didn´t try that out. It works, but is kind of unconvinient to my eyes.
Thanks for your help.

bsongis, 2020-06-05:
Bug confirmed

commit 0087c172 (2020-06-05): Fixes #7662
closed 2020-06-05 by bsongis