opentx/opentx #7663: Increase the MicroSD transfer speed on the USB port

robustini, 2020-06-03:
I honestly don't know if it's a memory problem or something else, but on all my TXs the speed of transferring files via USB is somewhat painful.
The problem increases in the sequential reading/writing of small files.
Now I'm reading from the X12S and the bitrate is clearly visible in the photo.
I tried with very fast MicroSD but the problem persists.
Obviously the MicroSD connected directly to the PC has its proper transfer rate.
Could something be done is a hardware bottleneck?

[image]

3djc, 2020-06-03:
Hardware, we are not slowing it down

closed 2020-06-03 by 3djc

robustini, 2020-06-03:
> 
> 
> Hardware, we are not slowing it down

So nothing can be done, too bad, thanks!

3djc, 2020-06-03:
Yes, I agree it is too slow. I actually remove it from the radio and read it directly on the PC for large transfer

robustini, 2020-06-03:
The problem is not in large files.
For example, I write the firmware.bin which is over 1 Mb in about 5 seconds.
The bottleneck is on sequentially writing very small files, but I understand that it is a hardware limitation, thanks anyway!