opentx/opentx #7667: Horus X10 with option INTERNALACCESS Simulator works with wrong Telemetry SubIDs

ParkerEde, 2020-06-03:
OTX 2.3.7:
The old X10 with old iXJT works with 3 digit SubIDs for telemtriesensors. OTX is okay so far.
The old X10 with ISRM Upgrademodule (with INTERNALLACCESS) works with 1 or 2 digits SubIDs OTX is not okay, because it still works with 3 digits of ID's

In relation of Telemtriesimulation the X10 with INTERNALACCESS must work like the X10 EXPRESS.

Kind regards
Ralf

ParkerEde, 2020-06-09:
cool, it's fixed since 2.3.8N239
Thank you
Ralf

kilrah, 2020-06-09:
What does that mean? in simulator only?

ParkerEde, 2020-06-09:
yes. In the physically radio is all fine. The issue is only in Companion radio simulator.

bsongis, 2020-06-10:
Please check the latest nightlies, this issue should be fixed (I tested just now). Feel free to reopen if I am wrong.

ParkerEde, 2020-06-10:
i've checked nightly 2.3.8N241. If a model is selected with ISRM D16, Telemetrie Simulator now is fine. If I choose a model with ISRM ACCEESS, only RSSI and RxBT works in Simulator. All other Sensors are still with 3 digits IDs and did not work.
One thing I've detected now in Simulator too. If you starts the Simulator and choose the "MDL" first Programmpage, and have a look to the Used RF Moduls. You can see, there is shown "OFF" for Internal RF, even though it was received from Horus as ISRM D16 or ISRM ACCESS.

bsongis, 2020-06-10:
[image]

[image]

[image]

bsongis, 2020-06-10:
[image]

[image]

ParkerEde, 2020-06-10:
Did you use in Profile a Horus X10 (old) with "internalaccess"?

3djc, 2020-06-10:
Simulators DO NOT take into account options. So you have a plain x10 simu, which cannot handle ISRM

bsongis, 2020-06-10:
Yes I did:
[image]

bsongis, 2020-06-10:
Ah yes we should include some other simulators, options are taken into account, but not all simulators with all options can be added to the installer bundle!

bsongis, 2020-06-10:
Will be solved once we have something to build on-demand simulators (like your firmware), but that's another story!

closed 2020-06-10 by bsongis

ParkerEde, 2020-06-10:
I found out, if I start simulator from Main WIndow with a default Model with ISRM D16 and switch inside the simulator to model with ISRM ACCESS then the Telemetrie will not work as it should. If I start the simulator from the Model editing window with a model uses ISRM ACCESS then it works like to see in your screenshots. It's okay for me but not perfect.
Many Thanks
Ralf