opentx/opentx #7668: Telemetrie programming page shows wrong ID's (n-1)

ParkerEde, 2020-06-03:
OTX 2.3.7
The physicaly Horus X10 with ISRM and INTERNALACCESS shows the SubID n-1 in the overview of telemetry programing screen. 
Example: In the Telemetrie overview, RSSI is shown as ID24. In the detailed RSSI Sensor programming page, there is the ID shown as 25 (this is the correct ID). 
Same for Simulator.

Kind regards
Ralf

RetoD7, 2020-08-19:
The Error still exists in OTX 2.3.9 on X10S Express.
(It's confusing when analyzing telemetry errors of FrSky RX ACCST 2.1.x and ACCESS 2.1.x: [wrong telemetry data with ACCST + ACCESS #83](https://github.com/FrSkyRC/Firmware-Test/issues/83) )

ParkerEde, 2021-02-16:
unstale

MarkusPfaff, 2021-05-20:
In OTX 2.3.11 on X9d+ as well as X12s the IDs shown on the telemetry page of the model settings differ from the correct physical ID shown in the detailed sensor value menu if you get the sensor table entries by "discover new sensors"

Examples:
**physical ID 21 is shown as 244**
**physical ID 25 is shown as 248**
This is on both Tx as well as their simulations in the companion.

If you need a complete list of all IDs I could generate one using a sensor which allows to set the pysical ID.

If you edit the telemetry tab in the companion and run the simulators then the numbers will be somewhat more correct in that they will be shown as n-1 (the behavior stated in the initial post in this issue.

MarkusPfaff, 2022-04-17:
The issue still is there.

closed 2022-11-04 by ParkerEde