opentx/opentx #7669: Open the .csv telemetry logfiles in the Log Viewer by double click

git2323, 2020-06-03:
It would be nice to be able to associate the .csv telemetry log files with the companion executable.
In this way we could open the telemetry log files (.csv File) by double clicking on them in the Windows Explorer.
Currently we have to start the companion, then open the log viewer, then Open Logfile, switch to the right path, select the file and click on 'Open'

Thanks

3djc, 2020-06-03:
Thats a configuation you can do for yourself in windows

3djc, 2020-06-03:
Ok, so probably what he would like is once you manually associate .csv to Companion, that companion can open log viewer with that file instead of main windows

git2323, 2021-03-14:
Would still be nice to have it ;-)

closed 2022-04-17 by stale[bot]