opentx/opentx #7673: 9XR PRO global variable and Logical switch do not work properly

Berriss71, 2020-06-04:
Hello,

By migrating from version 2.3.3 to higher, logic switches and special functions stopped working.

For example:
**Special features:**
SF1:! TRN modify GP2 +1
SF2: L01 modify GP2 value 0

**Logic switches:**
L01: a>x V1: GP2 V: 4 AND swintch N / A

If I simulate on OPET TX 2.3.3 everything runs as it should.
After 5 TRN switching, the GP2 value exceeds V2: 4 and is reset
In the higher version, this no longer works

if I am smelling the same on Horus X10, for example, everything is OK

 OpenTX version 2.3.7
 Hardware: Turnigy 9XR Pro

kilrah, 2020-06-04:
Likely fixed by https://github.com/opentx/opentx/pull/7639
Try a nightly build

Berriss71, 2020-06-06:
I tried nightly build 2.3.8N231 but it doesn't work yet. It's already 2.3.9 I can't get to it.

OpenTX version 2.3.7
Hardware: Turnigy 9XR Pro

aeronaut999, 2020-12-13:
Is this problem possibly related to this issue https://github.com/opentx/opentx/issues/7949 ?

closed 2021-09-22 by stale[bot]