opentx/opentx #7674: Increase the number of Special Functions

Mateyhv, 2020-06-05:
Hi, I know memory is scarce (X9D+ here) but is it possible to increase the number of Special Functions available? The reason is they do a lot of functions however for a very complex and customized models with sounds and tracks for all switches and many possibe situations, most of the SF are used just for playing tracks which in the end prevent using more advanced functions.
So would be useful if SF can be increased

kilrah, 2020-06-05:
We already doubled them recently - pretty much no chance more would be added especially not on the X9D+ since that's the one that has the least memory available.

But most of the play tracks can probably be moved away from SFs, that's what the model sounds are for

closed 2020-06-05 by kilrah

Mateyhv, 2020-06-05:
Thanks for the quick response Kilrah! Believe or not I was unaware of thr Model sounds, will make use of them where available to free Special Functions! Thanks