opentx/opentx #7675: Companion 2.3.8N229 - Error when importing a log

robustini, 2020-06-05:
I discovered now that I cannot open the logs generated with the current firmware compiled by the "2.3" branch, generated by Horus X10S Express.
The error is the one shown in the image.

[Epsilon_XL3-2020-06-05.zip](https://github.com/opentx/opentx/files/4735736/Epsilon_XL3-2020-06-05.zip)

[image]

**General information**
 - OpenTX version V2.3.8 (self compiled yestarday from 2.3 branch)
 - Hardware Horus X10S Express
 - OpenTX Companion V2.3.8N229

3djc, 2020-06-05:
Did you add/remove sensors ?

robustini, 2020-06-05:
No, but among the sensors there were none active, that yes.

robustini, 2020-06-05:
The same log works perfectly in the sim.

3djc, 2020-06-05:
You added data to the log between 10:12:13 and 10:35:51, and you can't do that

closed 2020-06-05 by 3djc

robustini, 2020-06-05:
> 
> 
> You added data to the log between 10:12:13 and 10:35:51, and you can't do that

That is, did I add sensors while the log was being written?
This is possible, and if this is the problem it really isn't, just know it! :-)

3djc, 2020-06-05:
Yes, it is a know limitation, you cannot change sensors been logged in a single log (day)