opentx/opentx #7676: Restore from *.OTX backup

sgmservice, 2020-06-05:
I had an issue yesterday with a Jumper T16 v2. I had finished a flight and was getting ready to go up again with a different plane. When I switch the radio on I got a “Storage Error”. 

When I cleared the error all of the models in memory and radio settings were gone. The radio seemed to be working fine but all of the models and radio setting were cleared, as if factory reset. All of the data appeared to be present and available on the SD Card, images, sounds, logs, backups, etc. 

To the best of my knowledge, there is no way to reload from an *.OTX backup on the radio itself and the setup for planes that I had at the field wee more complicated than I wanted to deal with there, so home I went. When I got home I was able to restore by simply pulling up a backup OTX file in Companion and loading it to the radio. Lo and behold everything is fine and dandy. 

I am not submitting the erasure of memory as an issue because I have NOT been able to recreate the error. I'm chalking it up to a software/firmware/hardware glitch.

A suggested new feature would be to have the ability to reload memory from an *.OTX backup file on the radio. This would give users the possibility of going back to a restore point when Companion is not readily available.
TIA, and BTW I love Open-TX, greatest thing since COBOL
Gene Maurice

Mateyhv, 2020-06-08:
You have the possibility to backup your models IN the radio, from the Models page. Same for restore a backed up model. Files go to the SD card MODEL folder if I am not mistaken.

kilrah, 2020-06-08:
That's correct for Taranis/BW screen radios but not for the T16 and other color screen ones.

sgmservice, 2020-06-08:
I apologize, I should have stated in the issue that I am referring to a Jumper T16, and other color screen radios.

kilrah, 2020-06-08:
You did :)

closed 2021-03-20 by stale[bot]