opentx/opentx #7681: Model specific audio files not playing when flight mode name longer than 8 characters

rwebb616, 2020-06-06:
This is a really odd one that took me a while to figure out what was happening.  

Current Behavior:  When a flight mode is created and set up with a name of greater than 8 characters, audio files in a model specific folder for switch configurations such as SG-up.wav no longer play.

Expected Behavior:  Changes in flight mode configuration settings should not affect the playback of model specific .wav files for switches etc.

Steps to reproduce (making sure that there is an audio file in the model specific folder:

1.  Create a new model
2.  Using radio or companion go to flight mode 1
3.  Name flight mode 1 a name longer than 8 characters (stabilize in my case) and save settings
4.  Load a different model
5.  Reload the previous model
6.  Attempt to toggle switches that should have sound files attached to them.

Not all switches in my case weren't working for example switch F worked on my radio but a logical switch and switch G were not working.

My hardware is Jumper T16 Pro v2.
Rich

3djc, 2020-06-06:
I'm unable to replicate this. Please incldue your OTX and SDcard content

rwebb616, 2020-06-06:
Do you want complete sdcard content or only changes from stock? And is OTX my firmware file? Also I'm going to go back to factory stock and try to reproduce to see if any additional steps are needed.

3djc, 2020-06-06:
Maybe try with this T16 firmware first

[firmware.zip](https://github.com/opentx/opentx/files/4740160/firmware.zip)

rwebb616, 2020-06-06:
It does work with that firmware.  Any idea what it is?

3djc, 2020-06-06:
Yes :)

3djc, 2020-06-06:
Should be ok in tomorrow’s nightly

closed 2020-06-06 by 3djc

rwebb616, 2020-06-06:
Mine still will not play P11.wav for 6P1 on the T16.  Am I naming that incorrectly?

3djc, 2020-06-06:
Could you try S21.wav, S22.wav for pos2,..

rwebb616, 2020-06-06:
That appears to work so maybe an update to the documentation?  Indicates P+Switch+Position.

Thank You! 
Rich

PR #7688 (2020-06-06): Sanitize AUDIO_FILENAME_MAXLEN
This fixes #7681