opentx/opentx #7686: Frsky QX7 Scroll Wheel jumping up after flashing to 2.3.7

GotskGK, 2020-06-06:
Hi there,

I've got a Taranis QX7 ACCST that I just flashed to opentx 2.3.7 using the companion and it flashed no problem. However, now, in any menu when I try to scroll down the scroll wheel will jump back up and the radio is becoming unusable. It looks like it's the same issue as this previous thread:

https://github.com/opentx/opentx/issues/5492

but despite the fact that I'm now using 2.3+ I've got the issue? I would roll back to a previous firmware version (as I didn't have this issue before flashing) but I can't find where the .bin file might be as the companion (even though I tried using an older version of the companion) always forces me to download 2.3.7.

Thanks.

GotskGK, 2020-06-06:
Just to add, if it's any help, I get this behaviour in the boot loader as well.

kilrah, 2020-06-06:
Must be something strange with the hardware, there are thousands of people on 2.3.7 with a QX7...

GotskGK, 2020-06-06:
Opened it up and there seems to be no issues with the rotary encoder itself (presuming that's what it is). Just to check though, is there any way I could have the firmware file for an older version of opentx? If that doesn't work I'll be replacing the scroll wheel. Thanks

3djc, 2020-06-06:
You can download OpenTX 2.2.4 using Companion 2.2

GotskGK, 2020-06-06:
Ah thanks for that. Flashed with an older version and clearly the newer version of opentx is not at fault. Sorry for wasting your time. Last question, I've looked around but haven't found anywhere where I can buy a replacement scroll wheel? Are there some generic ones that might fit? Thanks.

3djc, 2020-06-06:
Closing as this is not an opentx issue

closed 2020-06-06 by 3djc