opentx/opentx #7689: OTX 2.3.8 - X12S with ISRM: shows wrong selection (internal + external antenna)

robustini, 2020-06-06:
With the "internalaccess" enabled it should not show "internal + external antenna" but only "external", as it should be in case of IXJT module.

**General information**
 - OpenTX version 2.3.8 N231
 - Hardware Horus X12S with ISRM

[image]

MIKE-DAILEY, 2020-06-07:
The antenna line item does not exist in OpenTX-x12s-2.3.8N231 (f6ce40e7)  

It is a X12S internalaccess English.

kilrah, 2020-06-07:
The setting does not apply to ACCESS, needs to be removed.

MIKE-DAILEY, 2020-06-07:
Right and it is apparently OK with X12S internalaccess English

3djc, 2020-06-07:
This is because he is building himself and forget to have  INTERNAL_MODULE_PXX1 set to off

closed 2020-06-07 by 3djc

robustini, 2020-06-07:
> 
> 
> This is because he is building himself and forget to have INTERNAL_MODULE_PXX1 set to off

That's right, I added the option now, thanks!

robustini, 2020-06-07:
Sure, but I can show you a TMX5 that Horus X10S Express and Jumper T16 both flash perfectly, but X12S does not.