opentx/opentx #7692: Can't change widget colors

michifpv, 2020-06-06:
When i change the color of my telemetry widget to red the color stays the same (blue). Only for one value i can change it to red. All other colors are set to red but stays blue. Any fix for this?

Radio: Radiomaster TX16s with stock Firmware.
[image]

3djc, 2020-06-06:
Hello. This place is only a tracker for OpenTX issues, not a community support place. Please go to places like rcgroups or openrcforum for this type of question.

closed 2020-06-06 by 3djc

Marsellus19, 2020-10-15:
I have the same issue. The colour of the text can be changed for any displayed value, but not for telemetry data. I looks like either an openTX issue or the default Value widgets issue.

kilrah, 2020-10-15:
The color you choose will be used whenever you're actually receiving telemetry. When there's none it's always blue.