opentx/opentx #7693: R9M 2019 to R9MM bind problem.

veselin4, 2020-06-07:
X9D+(pre 2019) flashed with OpenTX 2.3.7 (opentx-x9d+-flexr9m-lua-noheli-sqt5font-en-2.3.7.bin) but I also tried without flexr9m option with different tx/rx firmwares.
EEPROM 2.3.7
R9M 2019 Module (sticker on module says 2019 and SmartPort only but not specifying ACCST or ACCESS) is currently flashed with 20190117 flex accst firmware.
R9MM/MINI receiver which i had for a while is currently flashed with 20190219 fport flex accst firmware.
Problem is after upgrading from an old TX firmware that I had the receiver bound and working it stopped working and would not bind.
Out of despair for the last 3-4 days I literally tried all possible firmware combinations available from FrSky (ACCESS,ACCST,FCC,LBT,with or without f.port even tried OTA firmware) and the closest I get to binding is I get telemetry back during the binding chirp, like it would say "low battery" and "telemetry recovered" but when i push the bind button again to store it or also tried pushing exit button it would disconnect and lose connection and resetting receiver or transmitter would not connect. It would say "telemetry lost" also the back of the module will switch from rapid blinking to just solid red after 1-2s of pushing the bind button to stop the chirping. Also the receiver will never go into a flashing red or green, it will be solinf red and green in bind mode and just solid red if powered normally. One of the firmwares i had on during testing would cause the rx green light to be always on in bind or normal mode and the red would not be on at all which while looking promising would not bind and there was no connection between.
The opentx firmware that was on the tx when they were bound and working was from a year ago and I am not exactly sure which version it was but believe it was 2.2.1 or 2.2.2. I was not updating tx for long time out of fear that something like that would happen or lose all my settings and models which was just 3 models.
So flashing over 100 times the last 3 days and i noticed so many of the r9mm  fw files were same in size but was flashing them anyway one by one and testing in between, then flash the next module fw and start over with all the rx firmwares, about 40 downloaded. Today I came up with the idea to sort them out and check them with a duplicate file checker that checks by their contents not just the file size and behold - only 8 different r9mm/mini file versions for ACCST from 20+ i was reflashing redundantly. I gave up on trying any more ACCESS versions as it looks like my pre 2019 x9d+ does not support it. As I also remember when I first started the r9 system an year ago I had no success with ACCESS fw.
So I wrote some questions on some rc groups, read a dozen articles and videos. All instructional videos have slightly different hardware combinations, but most of them use older opentx fw. I assume the 2.3.7 opentx does not support intentionaly my combinaion of tx+r9module+r9rx becouse of the old internal hardware or there is bug. As just a consumer I have no idea of how the open tx and hardware work together and I am not even sure if I am asking the right questions or if I am using the right logic to come to my conclusions.
I would like to ask if anyone on here is using 2019R9M module in their pre 2019 Taranis X9D+ ACCST 2.4GHz radio? Oh I also tried flashing the internal 2.4 module per a few online forum instructions, thankfully there were only 2 fw versions for it.
Also some or most of you would probably know how the r9m binding process works would you please explain it quick to me? The receiver and module exchange their IDs and store them onboard or just the rx stores the rx module id? or maybe even the tx module sends it over to open tx to store it in the model settings ?

Any help is appreciated, Thank You!

kilrah, 2020-06-07:
I just tried an X9D+ on 2.3.7, an R9M (non-2019) on 20190117 Flex set to transmit on 868 at 10mW in the menu and an R9 Mini on 20190201 Flex Fport. It needed a couple of tries but it bound and seems to operate correctly (don't have Fport capable hardware, but RX LED is green and telemetry is present). Maybe try a different "Receiver no" on the radio. 

Make sure your radio has Flex FW loaded and is set to the right mode (868 or 900, not EU/FCC)

Your radio can't do ACCESS so no point checking that. 

> The receiver and module exchange their IDs and store them onboard or just the rx stores the tx module id?

The RX stores the module ID, that's all.

veselin4, 2020-06-07:
My R9M module is 2019 version, I wonder if that has something to do with it. I did try all this multiple times and seems like it is half binding, radio will start receiving telemetry while in bind mode but receiver lights don't change - both solid on. Could you please make a video of how your receiver acts when binding? I think maybe my rx has gone bad by just sitting in the bin? Do you know if my rx was r9mm-OTA maybe that is the problem? I will try chosing different rx number in the radio but I think maybe I already tried that.

Highviz, 2021-02-24:
Did you resolve the binding issues?  I don't know if it is any help, but I have a variety of transmitters running both the 2019  R9M module and pre 2019 Modules on 20190117 flex accst firmware and am able to bind to R9mm, R9 and R9 slim rx's.  Two of the transmitters are running older firmware (2.2.4) and one is running 2.3.5 firmware.  So none actually on 2.3.7 !  I can swap a module between transmitters and as long as I select the appropriate Rx number in the model memory, it will retain bind.  If you haven't already done it, I would suggest reflashing your Rx firmware.  I have encountered problems with not getting telemetry on some combinations of firmware, even though I get successful bind, but that is to be expected if the firmware is not matched at both ends!

Lagarda, 2021-06-11:
> Did you resolve the binding issues? I don't know if it is any help, but I have a variety of transmitters running both the 2019 R9M module and pre 2019 Modules on 20190117 flex accst firmware and am [...]

Hi! Do you mean that the older SW works well? I have problem with binding R9M and R9 slim + OTA. They are not binding together. I tried different receiver's SW Flex, Fcc, LBT but no success. The control I have is Radiomaster TX12. Open TX site doesn't have older SW versions, so if you have links please point them. Thanks!

3djc, 2021-06-11:
I can bind no issue on tx12 with R9M. Make sure they are NOT running access firmware, it should work

Lagarda, 2021-06-11:
> I can bind no issue on tx12 with R9M. Make sure they are NOT running access firmware, it should work

Thanks! Do you have *.bin file which works fine with your HW?

Lagarda, 2021-06-12:
> I can bind no issue on tx12 with R9M. Make sure they are NOT running access firmware, it should work

THanks again! I downloaded old FW without ACCESS but ACCSS into both R9M and R9 slim + OTA. And now binded.