opentx/opentx #7694: OTA Issue

bsongis, 2020-06-07:
I just checked N231 on my X12S OTA with an Archer GR8 and N231 still has the issue with the “waiting” screen.  You select the GR8 file to flash OTA, select flash OTA and the “waiting” screen pops up.  You power up the GR8.  When the X12S detects the Gr8 the “waiting” screen should go away and display the GR8 to be flashed.  The “waiting” screen does not go away when the GR8 is detected, it stays displayed unit you press RET. I would think you would want this to work like it does on the non-color screen systems.

bsongis, 2020-06-07:
I have a X10 Express and a ARCHER GR8. I can try it myself

3djc, 2020-06-07:
Possibly related : #7684

MIKE-DAILEY, 2020-06-07:
Thanks for posting my note @bsongis.  If you need help testing I would be happy to help.

commit ec9de757 (2020-06-08): Fixes #7694
closed 2020-06-08 by bsongis