opentx/opentx #7696: Logic Switches not fully erased, affecting new switches on same position

Mateyhv, 2020-06-07:
Tested on latest Companion Simulator and OTX 2.3.7 on X9D+, file attached for test in the simulator. I have had this issue before with a deleted LS position and after being populated with new parameters, giving strange results as if some parameters of the previous switch are still in the code and messing up with the new one. Same new LS moved or created on a fresh never used before LS position works OK.

In the example provided there are only three switches involved. Of them L36 and L46 are the reused LS positions (sorry I don't recall what was there before) :
L03 -> arming: throttle low and SF down.
**L36** -> checks VFAS > 16,50V when disarmed (!L03), delay 8s
**L46** -> instant EDGE when L36 true (used for reseting flight data)

Expected behaviour is when L36 is true L46 becomes true for a split second and thats all. What happens is whenever L46 becomes true it resets L36 either so after 8 seconds the process repeats again and again. This also affects LS10-15 which are driven by CH11 and not related at all.

If I move or create switches L36/L46 to fresh never used before positions like L52/L53 they work as expected, no issues at all.

Thanks

[OpenTX_Github.zip](https://github.com/opentx/opentx/files/4742043/OpenTX_Github.zip)

kilrah, 2020-06-07:
And it you use the "Clear" in the same position?

Mateyhv, 2020-06-07:
I used "Delete" if thats what you mean. I also tried to copy and paste from a good LS position to the "bad" one and the malfunction persists, so it doesn't matter if the LS is created or copied on that position. 

And if a logic switch on that "bad" LS position is copied and pasted elsewhere, the wrong bits move with it to the new position.

lshems, 2020-06-08:
Be careful on how to 'fix' this. A lot of people rely on restoring the settings of a line by reassigning a switch/input after having set it to no switch.
And no, it's not just me.

Mateyhv, 2020-06-08:
Update: after messing a while with the model LS52/53 started behaving oddly. Deleted and created them again on **LS54/55**, worked OK so loaded the model into the Taranis and worked flawlessly in the field. Today I did some organizing of Special Functions, changed a few Global Variables and created two new LS on positions 56/57, all of that non-related to **LS54/55** and I have not touched them at all. During my testing on the simulator I noticed that those LS are behaving oddly again.

To be honest I am a little bit lost now. I am not sure what makes the LS go wrong. Attached is the model with the changes.
[OpenTX_Github_2.zip](https://github.com/opentx/opentx/files/4747238/OpenTX_Github_2.zip)

Mateyhv, 2020-06-13:
@kilrah, I think I have found whats going on, not related to erasing LS at all.

L36 -> checks VFAS > 16,50V when disarmed (!L03), delay 8s
L46 -> instant EDGE when L36 true 
SFxx -> Reset flight triggered by L46 true

I asumed "reset flight" just resets timers and telemetry but aparently it also resets logic switches, so they start counting again and the above sequence goes into a never ending loop.

If that is the expected behaviour please don't hesitate to close this issue. Thanks

kilrah, 2020-06-14:
It makes sense yes, I actually expected it would be related but didn't have time to work through the logic.
Not looked in detail, but you might be able to add a Sticky to store the change past the reset until some other condition like a Timer switch. It might get reset by the Reset Flight too though, so you might want to do ResetTelemetry and Reset Timers instead.

closed 2020-06-14 by kilrah

Mateyhv, 2020-06-14:
As you said Sticky gets reset by Reset Flight but using Reset Timer + Reset Telemetry instead did the job!
Many thanks!