opentx/opentx #7697: Refresh button for Compare Models in Companion

Mateyhv, 2020-06-07:
Currently if a Compare Models window has two models open and changes are done to any of them you have to close the Compare Model window, open it again and drag both models. Would be useful a simple refresh button which just reloads the models to the changes visible instantly.

Thanks

elecpower, 2020-06-14:
@Mateyhv this feature has been asked for a number of times and I agree it would be nice but the compare window only receives a copy of each model's configuration with no link back to the source so Refresh would require major re-engineering.
To complicate matters the source for each model could be from different model files and you could move or delete one of the models or close a model file during a compare and thus a refresh would produce unpredictable results.

Mateyhv, 2020-06-14:
Thanks @elecpower, I can see the related problems now. Closing the issue.

closed 2020-06-14 by Mateyhv