opentx/opentx #7698: 2.3.7 bootloader does not allow connection to OpenTX companion in Mac OSX Catalina

valerioa, 2020-06-07:
**Describe the bug**
I have updated firmware and bootloader on my Taranis XD9+ to 2.3.7. Since the update I cannot connect to it with OpenTX companion.
df-util cannot list my Taranis, after I turn my Taranis on, set it to bootloader mode and connect via USB.

```
/Applications/OpenTX\ Companion\ 2.3.app/Contents/Resources/dfu-util -l
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/
```

And OpenTx Companion complains that there are no transmitter connected.
In the syslog I see the following

```
sudo dmesg |grep Taranis
USB device 0483572014300000 - FrSky, FrSky Taranis Bootloader - detected termination of interfaceNub, currentPowerState 1, provider is non-NULL
```
Interestingly, if I connect the Taranis to my Macbook Pro, with the Taranis turned off, I get the correct output from df-util

```
/Applications/OpenTX\ Companion\ 2.3.app/Contents/Resources/dfu-util -l
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found DFU: [0483:df11] ver=2200, devnum=10, cfg=1, intf=0, path="20-3", alt=3, name="@Device Feature/0xFFFF0000/01*004 e", serial="207F36803548"
Found DFU: [0483:df11] ver=2200, devnum=10, cfg=1, intf=0, path="20-3", alt=2, name="@OTP Memory /0x1FFF7800/01*512 e,01*016 e", serial="207F36803548"
Found DFU: [0483:df11] ver=2200, devnum=10, cfg=1, intf=0, path="20-3", alt=1, name="@Option Bytes  /0x1FFFC000/01*016 e", serial="207F36803548"
Found DFU: [0483:df11] ver=2200, devnum=10, cfg=1, intf=0, path="20-3", alt=0, name="@Internal Flash  /0x08000000/04*016Kg,01*064Kg,07*128Kg", serial="207F36803548"
```
But, still OpenTX companion cannot connect to it

**General information**
 - OpenTX 2.3.7
- Bootloader 2.3.7
 - Hardware Frsky Taranis XD9+
 - OpenTX Companion 2.3.7 on Mac OSX 10.15.5 Catalina

valerioa, 2020-06-07:
I figured it out. If I have the sdcard insert into the Taranis, OpenTX Companion 2.3 will not connect to the Transmitter. However, if I unplug the sdcard, OpenTX Companion 2.3  will connect fine. Not sure this is the intended behavior, but this is how it works currently on Mac.

kilrah, 2020-06-07:
Dfu is used when connected with power off, not in bootloader mode.

valerioa, 2020-06-07:
Got it. Still, if I have an sdcard inserted, the Mac in unable to see the 'Taranis' virtual disk where the settings are. Thus, Companion cannot see it.

closed 2021-03-20 by stale[bot]

saintofinternet, 2022-06-11:
I am facing this problem since the latest upgrade of Firmware and Bootloader to 2.3.15.