opentx/opentx #7700: Companion not saving JP4in1 settings (v 2.37)

deaconFlyer, 2020-06-07:
Almost every time CompanionTX is resetting the DIY Multiprotocol (JP4in1) Multi-radio Protocol back to Hisky regardless of what I have saved it as. Normally the setting for me would be SFHSS.
- The custom Failsafe settings are also disappearing (with every save).
- I have to open each model, verify / reset the Protocol and the Failsafe, leaving them open in Companion until I do a write to the transmitter in order to get SFHSS to be transferred.
- Occasionally the Protocol setting will remain but never the Custom Failsafe. Next time, both will not be present. Trying to pin down exactly how/why.
I am using Companion on an Ubuntu (v20.20) system.

3djc, 2020-06-07:
Please try 2.3.8N232 and let us know, but it should be fixed

deaconFlyer, 2020-06-07:
Will do ... just to add more ... loaded my version from an hour ago ... SFHSS was there. Changed one of the Special Functions and just for fun checked the Setup again, it was back to Hisky.

deaconFlyer, 2020-06-07:
2.3.8N232 ... the Protocol is saving. The Custom Failsafe numbers are not.

3djc, 2020-06-07:
Could you please give more details then ?

deaconFlyer, 2020-06-07:
I always use a custom Failsafe which will drop the throttle (-100) and depending on the plane / heli will do its best to generate a soft landing (activating stablizer modes etc.). When I reopen CompanionTX and recall my OTX file, the failsafe values are all back to "0". (In 2.3.7 often the setting for the Failsafe Mode was initialized upon recall ... not stored at all (No Pulses/Receiver/Custom) ... so this is somewhat better.) Thanks for the help!

deaconFlyer, 2020-06-10:
Using N236 and the values seem to be holding. Thank you! Will continue to watch.

closed 2020-06-10 by deaconFlyer