opentx/opentx #7708: OTX 2.3.8 N240 - Volume sequence inside "Special Function" not respected at boot

robustini, 2020-06-09:
I noticed this a little annoying thing at boot, that in the X12S having two loud enough speakers is not nice.
If I set the volume control as the first "special function" as in the image and then I inserted the reading of wave files associated with logic switches, OpenTX plays part of the first audio file at the maximum volume of the main radio settings, and shortly after takes the value of S2 (in my case in the central position, so volume at 50%) by executing the subsequent audio messages at the volume of the S2 value.
It would instead be more logical for him to read first the value of S2 because it is the first on the list, and only then execute the subsequent audio events at that value.
Obviously this thing does it only on boot, but it's annoying.

**General information**
 - OpenTX version 2.3.8 N240
 - Hardware Horus X12S

[image]

Mateyhv, 2020-06-09:
Same happens on X9D+ having S2 configured as first Global Function. On boot the hello.wav which is about 8seconds track is played for a couple of seconds at 100% voumel and just then decreases to what S2 dictates. Not a big deal but prevents a 'silent' boot if necesary.

robustini, 2020-06-09:
> 
> 
> Same happens on X9D+ having S2 configured as the unique Global Function. On boot the hello.wav which is about 8seconds track is played for a couple of seconds at 100% voumel and just then decreases to what S2 dictates. Not a big deal but prevents a 'silent' boot if necesary.
> 
> Edit: at least in my case not exclusive to 2.3.8, has been happening forever.

This is exactly what has been going on for a long time.
On boot it would be better if the hello.wav file was reproduced at the volume setting of the selected model (if present in the special functions), but I would be satisfied that at least what I suggested above happened.
Of course, it's not a problem of 2.3.8, but testing that you have to leave a reference to the developers.

kilrah, 2020-06-09:
The hello.wav is played before the model data is loaded and the mixer runs, so no mode data can affect it. Just set your volume in system to an acceptable level since that's what's used until the model data is loaded.

Mateyhv, 2020-06-09:
Being so @robustini I guess the issue can be closed, what do you think?

robustini, 2020-06-09:
Yes, it is a workaround but it is not the solution, I mean on a logical programming level... :-)

robustini, 2020-06-09:
You can also create a completely silent hello.wav audio file, lasting the boot sequence, but not everyone can do it.

kilrah, 2020-06-09:
Again just set the main volume to 0 in the radio settings.

closed 2020-06-09 by robustini